ct-Open-Source/tuya-convert #935: How to reflash bulb with other firmware

hydazz, 2021-04-15:
I just successfully flashed my GLD112HA (Arlec Smart Globe) with ESPHome, forgetting to change the yaml to add any output or light so it's uncontrollable, the light is showing up in home assistant as `ESPHome: temp_flash` and is pingable, but I don't get how to put the light in the `pairing`? mode in order to flash my revised firmware with proper light controls. I wouldn't think that the light is bricked if it's showing up in home assistant and connected to wifi?

temp_flash.yaml (used to compile esphome):
```
esphome:
  name: temp_flash
  platform: ESP8266
  board: esp8285

wifi:
  ssid: "..."
  password: "...."

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Temp Flash Fallback Hotspot"
    password: "...."

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

```

hydazz, 2021-04-15:
problem solved by disabling my IoT wifi network and using the captive portal to flash new firmware!

closed 2021-04-15 by hydazz

AnthonyKNorman, 2021-04-15:
Do you have access to the EspHome add-on in Supervisor?