ct-Open-Source/tuya-convert #936: AP disables after few seconds on Kali 2021.1

nilvanis, 2021-04-15:
Hi,
I've encountered a problem running tuya-convert on Kali Linux 2021.1 and found why this is happening.

So when I run the start_flash.sh i can see all is setting up properly and for a brief moment I can see the vtrust SSID broadcasted and I can connect to it. After a few seconds it disappears and does not come back.
In the logs you can see:
```
(...)
Failed to restart network-manager.service: Unit network-manager.service not found.
(...)
wlan0: AP-ENABLED
wlan0: INTERFACE-DISABLED
wlan0: INTERFACE-ENABLED
Failed to set beacon parameters
```

The reason of this issue is that in Kali network-manager.service is running as NetworkManager.service.
Changing the two lines in scripts/setup_ap.sh to replace network-manager.service with NetworkManager.service solves the issue.

robmwalsh, 2021-04-17:
I had the same issue and I can confirm that this fixed it for me as well. Unfortunately it seems all my new devices have the new firmware so it was all for nothing :) time to start cracking things open :)

kueblc, 2021-04-19:
Thank you @nilvanis for sharing your solution! This would be a great enhancement to add to our scripts.

kueblc, 2021-11-13:
Resolved on `development`

closed 2021-11-13 by kueblc