ct-Open-Source/tuya-convert #940: Is it possible to downgrade firmware?

AleXSR700, 2021-04-24:
Hello everyone,
I own two Gosund SP112 devices and would love to switch to Tasmota, but I don't really want to open them and use the FTDI Mod.
I was wondering if it is possible to somehow downgrade the firmware back to a compatible version? Are the USB ports maybe connected or are they power supply only?

Thank you for your great work :-)

kueblc, 2021-04-24:
I'm afraid not, sorry. Tuya does not want folks installing custom firmware. The USB port is most definitely power only.

tarontop, 2021-04-28:
Devices with Factory Flashed Tasmota
https://templates.blakadder.com/preflashed.html

AleXSR700, 2021-04-28:
> Devices with Factory Flashed Tasmota
> https://templates.blakadder.com/preflashed.html

???

tarontop, 2021-04-28:
> > Devices with Factory Flashed Tasmota
> > https://templates.blakadder.com/preflashed.html
> 
> ???

Because the plug is not easy to remove, I directly bought the plug and bulb with Tasmota pre-installed

AleXSR700, 2021-04-28:
Ah, okay, but that is not really the solution I was looking for :-D

For all new devices I am going with Shelly in-wall solutions. But I would like to continue to use my existing Tuya plug-in switches.

t0ny-peng, 2021-08-21:
@AleXSR700 If you don't plan to use the smart plug to control a bank vault, you can try their official addon for Home assistant(sorry for those not using HA)

https://github.com/tuya/tuya-home-assistant