ct-Open-Source/tuya-convert #948: Unable to flash same batch of previously flashable devices

AndrewKCST, 2021-05-14:
Crew,

I was able to successfully flash a handful of Topgreener tgwf115pqm smartplug devices late last year. I have the same hardware revision devices and am unable to successfully flash devices of the same batch and am scratching my head as to why. This really does not seem like a #483 issue because the initial smartconfig completes, then firmware read stalls (do not care about original firmware backup FYI). Attaching some logs. Please help...
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/6476257/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/6476259/smarthack-web.log)

AndrewKCST, 2021-05-18:
So no bites. Here's the output of the flash attempt...

`======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.......................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
..Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
.................
SmartConfig complete.
Resending SmartConfig Packets
.............................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
`

AndrewKCST, 2021-05-18:
No bites? Well here is the output of the flash attempt...

`======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.......................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
..Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
.................
SmartConfig complete.
Resending SmartConfig Packets
.............................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
`

AndrewKCST, 2021-05-18:
No bites? Here is the output when attempting to flash...

`======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.......................
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
..Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
.................
SmartConfig complete.
Resending SmartConfig Packets
.............................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
`

phixion, 2021-07-31:
your device is already listed in https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02
as affected. It seems to be the error everyone's getting.

kueblc, 2021-07-31:
@AndrewKCST are you still having issues?

It looks like the device is an older Tuya firmware, pre TLS. What version of tuya-convert did you previously use that successfully flashed the last batch?

AndrewKCST, 2021-07-31:
@kueblc 

It was either 2.2 or 2.3. Was also using a 2018 version of Kali. I can't get the old versions to run because of depreciated libraries in the scripts. I'd be happy to try again with an older version of tuya-convert if I could actually get it to run propper. I've still got 4 of these plugs just collecting dust and would appreciate any guidance.

What is strange is that I have been able to flash other devices with the current version of tuya-convert successfully...

AndrewKCST, 2021-08-24:
@kueblc

Any ideas? I can send you these units. If you're willing, DM me...

hacker1024, 2021-09-23:
I'm getting the same issue here on Arch with my Kogan Smart Switch (KASPEMHUSBA). I ordered a few at once, and flashed one about a year ago. Now I'm trying to flash another one, but I get this result. It's been connected to Kogan's online service (which uses a rebranded version of Tuya Smart) for a while.

hacker1024, 2021-09-23:
Hmm, I switched from Arch to a live Linux Mint environment and it flashed with no issues.

hamiltron, 2021-12-18:
Linux Mint in VirtualBox also did the trick for me. Kali and Ubuntu timed out as noted above.

felipecrs, 2022-07-13:
Which version of Linux Mint?

hamiltron, 2022-07-14:
20.2 worked for me

skaterdude, 2023-01-11:
I have some kisslink and iselector smart plugs I got in 2018.  On the first round it looked like it was resending packets, but on subsequent tries it does what's posted here.  Does the latest version of this not handle old as well as newer devices?