ct-Open-Source/tuya-convert #952: Way for a Pool Heatpump?

MasterOfPuppets111, 2021-06-02:
Hello, I bought the following heat pump for my Pool. You can control it with the Tuya/Smart Life App. Do you think Tuya Converter will work with it? I have fear to test ist an can never go back... because it costs around 1200€...

https://www.poolex.fr/en/produit/2020-swimming-pool-heat-pump-poolex-silverline-fi

Best regards 
Pierre

javimurcia, 2021-06-09:
First of all, is that device supported by an open source firmware, or you know its gpio dessignations? because if you don't know that, you might be able to flash it with tuya convert to tasmota or any other firmware, but you won't be capable of doing anithing other than accesing the web interface.

And if you want to be on the absolute safe side, use serial flashing, and make sure to create a backup of the flash before programing. That way even if the new firmware bricks the esp chip, you will be able to reprogram it.

That being said, if its a 1200€ and you aren't sure of what you are doing you might be better with the stock firmware, and using a solution like the home assistant tuya integration, local tuya integration or TradeFace tuya gateway, that don't risk bricking the device.

MasterOfPuppets111, 2021-06-09:
Hello javimurcia,

Thank you for the answer,

That's exactly  the question... I don't know if tasmota supports pool heat pumps... tomorrow I get the device, first I want to look was microcontroller is inside.

Should I first ask the tasmota community if this device is supported by the firmware?

Best regards 
Pierre

javimurcia, 2021-06-10:
Well, in reality you are taking on 2 problems at once:

- Am i capable of flashing this device?
- Can a do something productive with the firmware i have flashed on the device?

The first issue is what tuya-convert handles, but since it appears that no one has thinkered with said device, your only bet would be to try it and see if it works. It can have the new PSK format (which would prevent tuya convert from working, at least for now) or have the new realtek chips (which will be a dead end, since tasmota can not run on non ESP chips, and so is the case of tuya-convert)

But being a 1200€ appliance, and with so many unknown variables, i would certainly go the serial flashing route as is almost bulletproof (as long you don't release the magic smoke via a wiring error, you can always restore the original firmware).

The second issue is what an open source firmware such as tasmota (but also esphome, espeasy, espurna or any custom firmware you could build yourself) takes care of. All those firmwares can run on any ESP8266 and derived chips, but running the firmware is not enought, you also need to make the processor control things in the real world.

And this is where the problem lies, the firmware is running on the device, now you need to tell said firmware what it needs to do. You need to tell the firmware that it has a temperature sensor (and of what kind) in pin x, and the heat pump is pwm controlled via the y pin, to create a thermostat using the temperature sensor and the heat pum pwm, and so on.

But wat, there is more, ESP chips have very little i/o (that measn the can't control many things at the same time) so most probably the tuya module only handles the iot aspect of the device, and communicates with the rest of the electronics via an uart link.

You can have a look at https://templates.blakadder.com/ where people uploads the configuration (called templates) for devices. If your devices isn't in said database, it doesn't mean it's unsuported, but you will have to reverse engeneer how the device works

But before you take al this work, take time to think if all this is worth it. Bassically, what do you want to acomplish, and why do you want to reflash said device?. If it's for privacy or local control, you can pair your device with the official tuya app, get the localkey, and then block all internet access to the device and allow only local control.

MasterOfPuppets111, 2021-06-10:
Thank you for your big answer!

I think you told me the right way. The Heat Pump is now in my trunk, the next steps I take:

1. Look what controller is inside:
If it is a Realtek Chip, the story ends here :(
If it is an esp, I go on with:

2. How does the controller work? do the esp all of the controlling (I dont think so), or communicate it with another controller and do only the IoT thing (if think like you that that is the probably way).

3. Thank I have to check if I could reverse engineering the commands on the serial port.

On Blackadder if found no template =(... I think that is much work....

I want to do this, because I have a knx installation in my home and want to get off the cloud thing to control it via a gateway in knx (edomi).

Thank you very much.

javimurcia, 2021-06-10:
Tuya has an api and a library for handling the comunication between the tuya module, and the device's mcu check https://tasmota.github.io/docs/TuyaMCU/ and https://images.tuyacn.com/smart/aircondition/Guide-to-Interworking-with-the-Tuya-MCU.pdf so you aren't in the dark.