ct-Open-Source/tuya-convert #957: Device Stopped Working after 2 days use

asosnovsky, 2021-06-21:
Device:  Prime Smart Outlet Outdoor Plug CCWFIO232PK

For some reason after 2 days of working just fine, it start started blinking red and is no longer on my wifi network. I tried re-running tuya-convert, and it seemed to have picked it up, but failed to re-flash.

```bash
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..............................
Timed out while waiting for the device to (re)connect
Stopping smart config
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
```

Logs 
```log
1624048394: New connection from 10.42.42.24 on port 1883.
1624048394: Client 44860572840d8e430b9b already connected, closing old connection.
1624048394: New client connected from 10.42.42.24 as 44860572840d8e430b9b (p1, c1, k60, u'44860572840d8e430b9b').
1624048394: Will message specified (107 bytes) (r0, q1).
1624048394:     tuya/smart/will
1624048394: Sending CONNACK to 44860572840d8e430b9b (0, 0)
1624048394: Received SUBSCRIBE from 44860572840d8e430b9b
1624048394:     smart/device/in/44860572840d8e430b9b (QoS 0)
1624048394: 44860572840d8e430b9b 0 smart/device/in/44860572840d8e430b9b
1624048394: Sending SUBACK to 44860572840d8e430b9b
1624048401: Received PUBLISH from 44860572840d8e430b9b (d0, q1, r0, m2, 'smart/device/out/44860572840d8e430b9b', ... (147 bytes))
1624048401: Sending PUBACK to 44860572840d8e430b9b (m2, rc0)
1624048406: Received PUBLISH from 44860572840d8e430b9b (d0, q1, r0, m3, 'smart/device/out/44860572840d8e430b9b', ... (147 bytes))
[... 462 lines cut ...]
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.8.5
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp4s0...
Configuration file: /dev/stdin
Using interface wlp4s0 with hwaddr f0:d5:bf:5f:53:e8 and ssid "vtrust-flash"
wlp4s0: interface state UNINITIALIZED->ENABLED
wlp4s0: AP-ENABLED 
wlp4s0: AP-STA-CONNECTED 36:b0:46:10:fc:6b
wlp4s0: interface state ENABLED->DISABLED
wlp4s0: AP-STA-DISCONNECTED 36:b0:46:10:fc:6b
wlp4s0: AP-DISABLED 
wlp4s0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlp4s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
```

kueblc, 2021-06-25:
If you've flashed the device and it is no longer running stock firmware, tuya-convert would not be able to help you there.

Try asking in the support channels for your alternative firmware, Tasmota, Espurna, ESPHome, etc.

asosnovsky, 2021-06-25:
Sorry I figured out the issue, I inputted two SSIDs for the network, and it was randomly switching between the two. I was able to fix this by going to the other vlan and updating the configs. Closing the ticket.

closed 2021-06-25 by asosnovsky