ct-Open-Source/tuya-convert #963: Issue with start_flash.sh

danielvega3009, 2021-07-19:
Hello,

I am flashing, I have followed the steps of tuya convert, but when I execute start_flash.sh the process stops and does not advance and the raspberry stops responding
[image]

How can I solve this?
thanks

kueblc, 2021-07-19:
Looks like you might be connected to your PI wirelessly. You must connect some other way.

See [Troubleshooting](https://github.com/ct-Open-Source/tuya-convert/wiki/Troubleshooting) for more information.

danielvega3009, 2021-07-19:
thanks, now my issue is that when starting smart config pairing procedure , the process ends with timeout meanwhile write in the terminal:
SmartConfig complete.
Resending SmartConfig Packets........

I connected my smartphone to the wifi vtrust-flash and I put my device in paring mode, but I got a timeout

kueblc, 2021-07-19:
It will attempt to automatically diagnose the issue after it times out. What was the message?

danielvega3009, 2021-07-19:
the message is Timed out while waiting for the device to (re)connect
=============================================
Attempting to diagnose the issue....
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.

For additional information, check the *.log files inside the scripts folder

Note: my device is a Tasmota smart plug based in esp8266

kueblc, 2021-07-19:
> For additional information, check the *.log files inside the scripts folder

Please share these logs

> Note: my device is a Tasmota smart plug based in esp8266

I assume you mean Tuya smart plug? Tasmota is already alternative firmware.

How have you determined that this device uses ESP8266?

danielvega3009, 2021-07-19:
Yes , my device is: 
https://www.athom.tech/blank-1/color-smart-plug

and is based in Tasmota.

In the logs:
smarthack-mqtt.log:
1626694147: mosquitto version 2.0.11 starting
1626694147: Using default config.
1626694147: Starting in local only mode. Connections will only be possible from clients running on this machine.
1626694147: Create a configuration file which defines a listener to allow remote access.
1626694147: For more details see https://mosquitto.org/documentation/authentication-methods/
1626694147: Opening ipv4 listen socket on port 1883.
1626694147: Opening ipv6 listen socket on port 1883.
1626694147: mosquitto version 2.0.11 running
1626694646: mosquitto version 2.0.11 starting
1626694646: Using default config.
1626694646: Starting in local only mode. Connections will only be possible from clients running on this machine.
1626694646: Create a configuration file which defines a listener to allow remote access.
1626694646: For more details see https://mosquitto.org/documentation/authentication-methods/
1626694646: Opening ipv4 listen socket on port 1883.
1626694646: Opening ipv6 listen socket on port 1883.
1626694646: mosquitto version 2.0.11 running
1626694810: mosquitto version 2.0.11 starting
1626694810: Using default config.

smarthack-psk.log:
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.12:45614
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1056)
don't panic this is probably just your phone!
^CTraceback (most recent call last):
  File "./psk-frontend.py", line 113, in <module>
    main()
  File "./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

smarthack-udp.log:
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667

smarthack-web.log:
Listening on 10.42.42.1:80
[32m[I 210719 22:29:13 web:2243][m 302 GET /gen_204 (10.42.42.12) 5.79ms
[32m[I 210719 22:29:14 web:2243][m 302 GET /mobile/status.php (10.42.42.12) 2.94ms
[32m[I 210719 22:29:14 web:2243][m 200 GET / (10.42.42.12) 4.88ms
[32m[I 210719 22:29:21 web:2243][m 200 GET / (10.42.42.12) 3.25ms
[32m[I 210719 22:29:21 web:2243][m 302 GET /favicon.ico (10.42.42.12) 3.22ms
[32m[I 210719 22:29:21 web:2243][m 304 GET / (10.42.42.12) 3.14ms
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80

smarthack-wifi.log:
61ba060
Linux vega 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Versión dnsmasq 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

Este software viene SIN NINGUNA GARANTIA.
Dnsmasq es software libre, y usted está autorizado a redistribuirlo
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
[... 62 lines cut ...]
Linux vega 5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020 armv7l GNU/Linux
OpenSSL 1.1.1d  10 Sep 2019
Versión dnsmasq 2.80  Copyright (c) 2000-2018 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus i18n IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC loop-detect inotify dumpfile

Este software viene SIN NINGUNA GARANTIA.
Dnsmasq es software libre, y usted está autorizado a redistribuirlo
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.8-devel
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.7.3
===========
Configuring AP interface...
RTNETLINK answers: File exists
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlan0...
Configuration file: /dev/stdin
wlan0: Could not connect to kernel driver
Using interface wlan0 with hwaddr b8:27:eb:83:c2:ba and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: AP-STA-CONNECTED e8:db:84:9c:76:23

kueblc, 2021-07-20:
There seems to be a misunderstanding. I am not sure what you are trying to accomplish.

tuya-convert enables you to install alternative firmware like Tasmota on an ESP device running Tuya firmware. If the device is already running Tasmota, you have no use for tuya-convert.

danielvega3009, 2021-07-20:
is it not possible to flash from one tasmota firmware to another custom tasmota firmware?

kueblc, 2021-07-20:
You can do that without tuya-convert. tuya-convert only helps you convert tuya. See Tasmota to learn how to update your firmware.

closed 2021-07-20 by kueblc