ct-Open-Source/tuya-convert #964: start_flash.sh and two other files corrupt after running install_prereq.sh

twshield, 2021-07-26:
On a Raspberry Pi 3B with the latest (2021-05-07-raspios-buster-armhf.zip) Raspberry Pi OS clean install as user pi.

Did
```
git clone https://github.com/ct-Open-Source/tuya-convert
cd tuya-convert
./install_prereq.sh
```
Both appeared to succeed.

Then going to run
`./start_flash.sh`
I get an exec format error.  Looking at this file it is binary data.  

File command reports it as data and 
taylor% hexdump start_flash.sh 
0000000 f457 0003 000c 0201 002e 0000 f456 0003
0000010 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000020 0000 0000 0000 0000 0000 0000 0000 0000
*
0000200 f457 0003 000c 0201 002e 0000 f456 0003
0000210 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000220 0000 0000 0000 0000 0000 0000 0000 0000
*
0000400 f457 0003 000c 0201 002e 0000 f456 0003
0000410 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000420 0000 0000 0000 0000 0000 0000 0000 0000
*
0000600 f457 0003 000c 0201 002e 0000 f456 0003
0000610 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000620 0000 0000 0000 0000 0000 0000 0000 0000
*
0000800 f457 0003 000c 0201 002e 0000 f456 0003
0000810 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000820 0000 0000 0000 0000 0000 0000 0000 0000
*
0000a00 f457 0003 000c 0201 002e 0000 f456 0003
0000a10 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000a20 0000 0000 0000 0000 0000 0000 0000 0000
*
0000c00 f457 0003 000c 0201 002e 0000 f456 0003
0000c10 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000c20 0000 0000 0000 0000 0000 0000 0000 0000
*
0000e00 f457 0003 000c 0201 002e 0000 f456 0003
0000e10 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0000e20 0000 0000 0000 0000 0000 0000 0000 0000
*
0001000 f457 0003 000c 0201 002e 0000 f456 0003
0001010 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0001020 0000 0000 0000 0000 0000 0000 0000 0000
*
0001200 f457 0003 000c 0201 002e 0000 f456 0003
0001210 0ff4 0202 2e2e 0000 0000 0000 0000 0000
0001220 0000 0000 0000 0000 0000 0000 0000 0000
0001230 0000 0000 0000 0000 0000 0000 0000
000123d

It appears to be repeating data.  Oddly it has the same size as the original file:
`taylor% ls -l start_flash.sh

-rwxr-xr-x 1 1000 1000 4669 Jul 24 13:34 start_flash.sh*`

Git status shows this file as modified:
        modified:   scripts/fake-registration-server.py
        modified:   scripts/setup_checks.sh
        modified:   start_flash.sh

but I did not check it was not corrupt after the git clone step, but I did check that a git clone on a Linux box gets a good copy of start_flash.sh.  And I checked that start_flash.sh also shows as data with the SD card from the RPi mounted on a linux box.

Looking at the other files that are marked as modified shows the same problem (both are original sizes too):

taylor% file fake-registration-server.py
fake-registration-server.py: data
taylor% hexdump fake-registration-server.py
0000000 0f3d e745 32e8 422f 140c 050c cf22 c560
0000010 ed84 023d b421 05fe e8e6 cec1 d2a2 870c
0000020 cd56 2206 860c 3216 2204 850c d216 3203
0000030 442f 140c 0d5d cf22 0560 ed82 2f32 6d45
0000040 0c02 5d14 220d 30cf 6f62 054a ed81 2f32
0000050 ed46 0c02 5d14 2d0d 050f ed80 2f62 5d4a
0000060 2102 fea1 3660 e020 204e e0c5 32e8 122c
(lines omitted)
taylor% ls -l fake-registration-server.py
-rwxr-xr-x 1 1000 1000 7564 Jul 24 13:34 fake-registration-server.py*

taylor% file setup_checks.sh 
setup_checks.sh: data
taylor% ls -l setup_checks.sh 
-rwxr-xr-x 1 1000 1000 5120 Jul 24 13:34 setup_checks.sh*
taylor% hexdump setup_checks.sh 
0000000 3136 6162 3630 6230 3538 3736 6165 6463
0000010 3230 3138 6433 3462 3133 3639 6337 6562
0000020 6639 3264 3433 3130 000a 0000 0000 0000
0000030 0000 0000 0000 0000 0000 0000 0000 0000
*
0000200 3136 6162 3630 6230 3538 3736 6165 6463
0000210 3230 3138 6433 3462 3133 3639 6337 6562
0000220 6639 3264 3433 3130 000a 0000 0000 0000
0000230 0000 0000 0000 0000 0000 0000 0000 0000
(lines omitted)

Don't know what is going on here, but I'm hesitant to run anything else from this OS install after this odd behavior.  I'm worried something malicious is going on.

kueblc, 2021-07-26:
Curious. I have never seen anything like this. Might your SD card be corrupt?

twshield, 2021-07-26:
I did not have any problems with running the OS, but an fsck of the filesystem shows lots of multiply claimed blocks.  So something made a mess of the filesystem.  Let me see if I have another SD card to try this again.

Thanks for the quick reply.

twshield, 2021-07-26:
Must have been the SD card.  New one sets up ok, but, of course, after all that the firmware in my new Lumiman bulbs is too new.

Sorry to bother you.

closed 2021-07-26 by twshield

kueblc, 2021-07-26:
Glad you found the root cause - sorry the tool didn't work out for you! Hopefully we'll find a workaround.