ct-Open-Source/tuya-convert #975: I cant see the vtrust-flash network!

VCTGomes, 2021-08-11:
I got the following error message

1628714754: mosquitto version 2.0.11 running
^C1628714762: mosquitto version 2.0.11 terminating
1628715181: mosquitto version 2.0.11 starting
1628715181: Using default config.
1628715181: Starting in local only mode. Connections will only be possible from clients running on this machine.
1628715181: Create a configuration file which defines a listener to allow remote access.
1628715181: For more details see https://mosquitto.org/documentation/authentication-methods/
1628715181: Opening ipv4 listen socket on port 1883.
1628715181: Opening ipv6 listen socket on port 1883.
1628715181: mosquitto version 2.0.11 running

==> ./scripts/smarthack-psk.log <==
    main()
  File "/home/kali/tuya-convert/scripts/./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
^CTraceback (most recent call last):
  File "/home/kali/tuya-convert/scripts/./psk-frontend.py", line 113, in <module>
    main()
  File "/home/kali/tuya-convert/scripts/./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

==> ./scripts/smarthack-udp.log <==
192.168.2.158 {"ip":"192.168.2.158","gwId":"7523347310521cf689a1","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyj979nf3q3theh","version":"3.3"}
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^CListening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
192.168.2.158 {"ip":"192.168.2.158","gwId":"7523347310521cf689a1","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyj979nf3q3theh","version":"3.3"}
192.168.2.38 {"ip":"192.168.2.38","gwId":"ebf881ac383c6bbd51on3q","active":2,"ablilty":0,"encrypt":true,"productKey":"keym9qkuywghyrvs","version":"3.3"}
WARNING: it appears this device does not use an ESP82xx and therefore cannot install ESP based firmware
192.168.2.171 {"ip":"192.168.2.171","gwId":"64288857f4cfa27fe2c6","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyf9fsmuufeywsr","version":"3.3"}
192.168.2.159 {"ip":"192.168.2.159","gwId":"04003617f4cfa2cc3fee","active":2,"ability":0,"mode":0,"encrypt":true,"productKey":"keyf9fsmuufeywsr","version":"3.3"}

==> ./scripts/smarthack-web.log <==
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
Listening on 10.42.42.1:80
Listening on 10.42.42.1:80
Listening on 10.42.42.1:80
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80

==> ./scripts/smarthack-wifi.log <==
Configuration file: /dev/stdin
Using interface wlan0 with hwaddr 2a:b2:b0:e2:ed:9e and ssid "vtrust-flash"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED 
wlan0: INTERFACE-DISABLED 
wlan0: INTERFACE-ENABLED 
Failed to set beacon parameters
wlan0: INTERFACE-DISABLED 
wlan0: INTERFACE-ENABLED 
Failed to set beacon parameters

What can I do

boozeman, 2021-11-06:
Check pull request: Try alternate NetworkManager-naming #994

I made those changes for scripts/setup_ap.sh script and now it work fine on Kubuntu 21.10

kueblc, 2021-11-13:
Hi @VCTGomes,

Did you try @boozeman's suggestion?

DevinCarpenter, 2022-03-05:
@boozeman Thank you! That did the trick for me. Should definitely in the codebase.

cannondale0815, 2022-04-22:
Yes, that did the trick! Thanks @boozeman !