ct-Open-Source/tuya-convert #978: Can't flash new Gosund SP111

SebLz, 2021-08-17:
Hello,

I'm trying to flash a recently purchased Gosung SP111 3450W power plug with an old desktop running Ubuntu 20 (with tp-link wifi dongle). After putting the plug in pairing mode and pressing "ENTER" to start flashing the device, the process will loop at "Resending SmartConfig Packets". The smart plug remains in pairing mode (fast blinking) and I have no entry in the wifi log, like it was not even trying to connect. I've noticed some exceptions in the PSK log when I connect another device (mobile phone or laptop) to the AP before hitting "ENTER". Have I missed something or are the new SP111 not flashable anymore (I've never paired mine with tuya application, it's just got out of its box) ?

[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7002806/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7002807/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7002808/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7002809/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7002810/smarthack-mqtt.log)

```
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N]
```

SebiPanther, 2021-09-14:
Same here with Gosund SP1

kueblc, 2021-11-13:
Hi @SebLz 

Since you weren't able to pair the device using EZ config, you could try AP config. Try #966 if you are interested and let us know if it worked for you.

SebLz, 2021-11-13:
> Hi @SebLz
> 
> Since you weren't able to pair the device using EZ config, you could try AP config. Try #966 if you are interested and let us know if it worked for you.

Thanks for the hint @kueblc, unfortunately I can't test this anymore since I flashed the device using the usb to serial method a while ago.

kueblc, 2021-11-13:
Great, I'm glad you were able to get your device running your desired firmware.

closed 2021-11-13 by kueblc