ct-Open-Source/tuya-convert #981: start_flash.sh fails with "ImportError: No module named 'Cryptodome'

gooman-uk, 2021-08-28:
Here's the output - any ideas?

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/openhabian/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/openhabian/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ImportError: No module named 'Cryptodome'
........................................................................................................................
Timed out while waiting for the device to (re)connect
```

I've tried "pip install pycryptodome" but that didn't resolve it.

caseyhofford, 2021-09-02:
I had to do `pip3 install pycryptodomex`

aruhri, 2022-12-22:
I had the same issue on Ubuntu and resolved it with `sudo apt install python3-pycryptodome`.

OracPrime, 2024-04-17:
Also a problem on Raspberry Pi 4 B running debian.  Also fixed (well the error went away, still nothing connecting yet) by sudo apt install python3-pycryptodome