ct-Open-Source/tuya-convert #982: nl80211 not found. AP mode not supported! Please attach a WiFi card that supports AP mode.

albilia, 2021-09-12:
Hello all : ) 

I installed Tuya Convert on docker synology according to the following instructions:
git clone https://github.com/ct-Open-Source/tuya-convert
cd tuya-convert
docker build -t tuya:latest .
Setup docker-compose:
then, I changed the name of docker-compose.sample.yml to docker-compose.yml.
The location of the docker-compose.yml file is:
/docker/tuya-convert/docker/docker-compose.yml
In the following route there are the following three files:
/docker/tuya-convert/docker/
a. bin
b. data
c. docker-compose.yml

The settings in the docker-compose.yml file are: https://imgur.com/QY5GZTC
All the files and their layout in the Tuya Convert folder are: https://imgur.com/a/K9lMHNq
I have no idea where I am wrong, and if I should guess; Apparently I did not get the idea right with the Smartphone Wifi.
I am close to completing the installation and only this matter has been following me for a few days. I would greatly appreciate any attempt to assist me with the matter.By the way, I can't see logs in any folder.
Thanks in advance, Nissim.
[image]

TotalCoolness, 2021-09-15:
I think maybe your synology wifi card is not supported.

albilia, 2021-09-15:
Hi @TotalCoolness
Thanks for the feedback, so that's the point ?! :) 
I'm pretty noob and not familiar with tamosta & tuya-convert. I had concerns that a network card was indeed needed for that matter. At the moment, however, I do not have a wireless network card installed on the Synology.
It seems to me that I made a big mash between the 'need for a physical network card' and the Hostapd Container.
https://registry.hub.docker.com/r/jsiu/hostapd/
Still can't figure out if I can create a virtual AP, or I will have to purchase a network card anyway, or if my Asus router settings can be used in some way.
I would be happy if you could please clarify this point to me.
Thanks in advance.

TotalCoolness, 2021-09-15:
Well I'm no expert myself and got a few issues with it.

I take it you don't have a raspberry pi ? Or maybe you've got a pc with a usb wifi card then install raspbian os in a virtual machine and try that.

albilia, 2021-09-17:
Hi @TotalCoolness
In the end I was able to set up Tuya-Convert through a YouTube guide.
Everything went smoothly until the moment I was supposed to connect via the iPhone to vtrust-flash ssid.
I did it late and probably because of that I turned the smart switch to Break.
And now .. 'new problem' to Learn how to get out of it :)
I think I should take a break with this matter, I have devoted a lot of time to it. Thank you very much anyway.
https://www.youtube.com/watch?v=oTXI42X2H10&ab_channel=EpicLPer

TotalCoolness, 2021-09-17:
Yeah thats the guide I saw when I commented. I should have included the link but didnt know if you had a pc to do it in.

closed 2021-09-19 by albilia