ct-Open-Source/tuya-convert #983: If you can't wait, use dev branch for arm (docker) devices.

albilia, 2021-09-12:
Hi @Daviteusz

I tried to install tuya convert in two ways on docker synology. In the first installation: git clone https://github.com/ct-Open-Source/tuya-convert
Retrieved the following error:
nl80211 not found. AP mode not supported! Please attach a WiFi card that supports AP mode.
however, when I did it the way you suggested:
git clone -b development https://github.com/ct-Open-Source/tuya-convert.
I received an indication of installing additional plugins that were not installed in the first option, it seemed to me optimistic until the moment I received the following error message at the end of the installation: 
admin@Synology:/volume1/docker/tuya-convert$ sudo docker-compose up -d
Password:
WARNING: The WLAN variable is not set. Defaulting to a blank string.
WARNING: The AP variable is not set. Defaulting to a blank string.
WARNING: The GATEWAY variable is not set. Defaulting to a blank string.
WARNING: The LOCALBACKUPDIR variable is not set. Defaulting to a blank string.
Creating tuya-convert_tuya_1 ... error

ERROR: for tuya-convert_tuya_1  Cannot create container for service tuya: create .: volume name is too short, names should be at least two alphanumeric characters

ERROR: for tuya  Cannot create container for service tuya: create .: volume name is too short, names should be at least two alphanumeric characters
ERROR: Encountered errors while bringing up the project.
I would love to know if you have a compiled file in favor of installation on docker synology that can run smoothly without any glitches.
Thanks in advance : )

closed 2021-09-13 by albilia

Daviteusz, 2021-10-02:
@albilia      

Setup docker-compose:

    you may adjust this docker-compose.yml, if necessary:
        environment-variables may be different, for example network-adapter may be different from wlan0
        adjust the volume folder, where you want your backups stored

albilia, 2021-10-03:
@Daviteusz
thank's a lot : )
In the meantime I installed Tuya Convert on Vmware Ubuntu.
I installed the Access Point driver on Ubuntu, and it works perfectly.