ct-Open-Source/tuya-convert #1004: Failed to flush FLHS-ZN04

deiger, 2021-11-17:
Previously I managed to flash a similar device, so it's likely a new firmware issue (the FCC ID remained the same, if that is to be trusted).

Logs:
```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
.Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
.................
SmartConfig complete.
Auto retry in 3s. ................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...
```
Logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7555154/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7555155/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7555156/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7555157/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7555158/smarthack-wifi.log)

Thank you for an amazing tool!

kueblc, 2021-11-17:
I see evidence of two different devices in your logs. Did you try multiple devices in the same run?

In your web log, there's a pre-TLS device, and I can see it downloaded the intermediate firmware.

In your psk log, I see a PSK v1 device, but it doesn't appear in the web log.

In your wifi log, I see three devices, most likely one is your phone or donor device.

infocfc, 2021-11-19:
Are you by chance using a fresh or upgraded since you succesfully used it last time, Raspberry Pi Image for Tuya-Convert? There's increasing evidence that tuya-convert in some cases doesn't work properly anymore on those, and your symptoms match that.

deiger, 2021-11-19:
This was a fresh install, on a clean pi image. I've long deleted the image I previously used.

I just deleted the directory and tried again, twice on the same socket, and it continues to fail:

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7571559/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7571561/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7571562/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7571563/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7571564/smarthack-wifi.log)

infocfc, 2021-11-22:
That fits. Maybe try docker or another platform. There are too many reports of tuya-convert no longer working on current Raspi images to ignore them.

kueblc, 2021-11-22:
That's good to know @infocfc. If anyone has time to track down what changed in recent RPi, I will gladly accept a PR.

deiger, 2021-11-23:
I just managed to successfully flash these through docker on a Raspberry Pi 4. The docker fails to run on Raspberry Pi Zero, because it's arm/v6.
Thanks for the help!

closed 2021-11-23 by deiger