ct-Open-Source/tuya-convert #1007: KMC 4 Outlet "Timed out while waiting for device to (re)connect

chris948, 2021-11-28:
I searched, but I don't see this specific issue.

KMC 4 outlet smartplug, Model 30608, looks like it was made in 4/2019. 

inside the device: TYWE3S, Model 30401

I have tried 2 different wifi adapters, internal and USB. Exact same issue. 

my phone is 10.42.42.26, 1e:da:e0:d3:d6:fc. The other two are the smartplugs that I've had sitting in a drawer for over a year. I had no problems doing this with the same hardware back in I think early 2020.

Uploaded the logs, the web logs seem to say things should be working.

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7614084/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7614085/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7614086/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7614087/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7614088/smarthack-psk.log)

chris948, 2021-11-28:
Hello, using a new vanilla ubuntu 21.10 obviously had some issue that I don't see.

I tried a ubuntu 20.10 VM inside windows10 with the same usb wifi module and it worked perfectly. 

Closing this because it's fixed, but if I can do anything to help identify a problem in a ubuntu OS, glad to help.

closed 2021-11-28 by chris948

kueblc, 2021-11-28:
Hey @chris948, glad you were able to resolve your issue quickly on your own.

I have seen a number of reports of incompatibility with more recent Ubuntu and RPi distros. One suspect change is the renaming of network-manager to NetworkManager. This was fixed by #994 merged into `development` but not yet on `master`.

Did you try using the `development` branch?

chris948, 2021-11-28:
Hey, thanks for the response. Appreciate the work here. 

I did have to fix the NetworkManager references, I poorly tried to reference that at the bottom of my post. So that improved the situation, but I still got those timeouts. That allowed my phone to connect, so that did work, but the timed out issue is after the fix. 

Sorry about that. That wasn't clear and that's on me.