ct-Open-Source/tuya-convert #1009: Nedis WIFIPO110FBK fails to flash

bphermansson, 2021-12-06:
I'm trying to flash a Nedis WIFIPO110FBK but it fails. It just keeps "Resending SmartConfig Packets" until it times out.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7661512/smarthack-mqtt.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7661513/smarthack-wifi.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7661514/smarthack-udp.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7661515/smarthack-psk.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7661516/smarthack-web.log)
[image]

watkef, 2021-12-09:
I've got the same isue, only with the Nedis WIFIP130FWT. Resending SmartConfig  Packets until it times out. It says "Attempting to diagnose the issue..." until it says my device does not use an ESP82xx. According to a website with an overview for Tasmota compatible devices, the WIFIP130FWT does indeed have a ESP8266.

[image]

watkef, 2021-12-10:
Not sure how I can see PSK ID in smarthack-psk.log to check if it has patched firmware, but [this](https://github.com/blakadder/templates/issues/1230) suggests the newer tuya products from Nedis have a WB3S module instead of ESP8266.

kueblc, 2021-12-10:
@bphermansson I found this in your web log:
> ModuleNotFoundError: No module named 'tornado'

The python based web server can't find a dependency. Did you run `./install_prereq.sh`?

kueblc, 2021-12-10:
@watkef I'm afraid you do, in all likelihood, have a non-ESP device. The integrated check is pretty robust, at least in terms of false positives.

bphermansson, 2021-12-16:
I think I fixed the lack of tornado, but I'm not sure. I was using a Fedora system and couldn't run the install script. But now I am on an Ubuntusystem and will give it another try.

bphermansson, 2021-12-16:
Now I got another answer, so now I now the problem:

> Attempting to diagnose the issue...
> Your device's firmware is too new.
> Tuya patched the PSK vulnerability that we use to establish a connection.
> You might still be able to flash this device over serial.