ct-Open-Source/tuya-convert #1013: Error when starting flash

rubyrose96, 2021-12-10:
Tried to start the flashing a few times, completely wiped and started again with a new install but I keep getting this same error when the flashing starts.
[image]

The script then proceeds to not find any devices.

pickup7, 2021-12-11:
Agree. Happens since around one week. Working with the latest Pi OS release on two identical Pi B+ 1 from 2014.  Old release updates complained when trying to install screen. So i migrated to latest. 
The "illegal instruction" or "ungueltiger maschinenbefehl" is probably due to a call of a binary executable not fitting the processor architecture. no clue which component or module causes the error but may well be connected with python 3 ? 
Anybody has an advice on how to debug the script ?
Thank you indeed.

schnurrzz, 2021-12-13:
I've got the same problem. Always Line 138 after starting number of illegal instruction  is changing everytime. Didn't find anything in the logs. 
My girlfriend purchased some "Crown led smart bulb" from amazon, after reading about smart life / tuya app I ended up here with a fresh rasp ios on my ten year old rasp 1 and eager to get rid of this chinese spy firmware. Using their apps to connect to my homenetwork is no option. Not sure if the bulb is using an espchip or maybe I'm lost because they already come with a rtl8710. 
[https://www.amazon.de/CROWN-LED-Gl%C3%BChbirne-Warmwei%C3%9F-Beleuchtung/dp/B08KFPF4NG/ref=pd_lpo_1?pd_rd_i=B08KFPF4NG&psc=1](url)

Their homepage is also great... 
[https://crown-led.de/](url)

schnurrzz, 2021-12-16:
Seems to be a problem of the pi os.
 
I started over with Virtual Box and a fresh linux mint vm installation on my ryzen desktop pc. After some problems getting my wlan stick to work in the vm I was able to run the tuya convert script with a stable vtrust-flash and without the line 138 error / illegal instruction. 

After all it didn't help much in my case, the script ran and resend over and over again. Just one time the resend took quite long and the blinking of the pairing mode changed to a longer intervall. Almost thought I got the bitch now. :D

In one other case the pairing mode of the bulb was different right at the beginning (a little longer intervall between the blinking as normal and i saw a smartlife wlan suddenly show up on my phone. I was wondering if the bulb got two different pairing modes?

However use sth. else than the pi os and u should be fine :)

schnurrzz, 2021-12-16:
Yeeehhaaa, I altered fake-registration-server.py from 20 to 10 and that made it work after some trys. Holy Shit.

schnurrzz, 2021-12-16:
So after I thought, it didn't work with the flash because there was no msg that the flash was successful of the Crown LED E27 4W 360 lm warm white it did infact work and flash tasmota on it. At first I thought it would be bricked because the light was off and it was my first time using tuya-convert ever, but the tasmota gui showed up to set wlan and after doing that and connecting to the ip of the bulb I finally got into the settings menue to figure out the configuration. 
I searched for a similar bulb template after not finding one for the exact name and this didn't work. 
So I checked the info to see what chip is exactly in my bulb and it's a ESP8285. 

I started to google for a template suitable for this chip and this one made the bulb work again: 
https://templates.blakadder.com/DMP-L1.html
Although my bulb isn't for color mode only the second one worked;
{"NAME":"DMP-L1 Light","GPIO":[1,1,0,1,419,418,0,0,416,420,417,1,1,1],"FLAG":0,"BASE":18}

Let's see if it is also fine for Alexa now.

Thanks to all who made this tuya-convert script, Merry Xmas :)

pickup7, 2021-12-17:
> Agree. Happens since around one week. Working with the latest Pi OS release on two identical Pi B+ 1 from 2014. Old release updates complained when trying to install screen. So i migrated to latest. The "illegal instruction" or "ungueltiger maschinenbefehl" is probably due to a call of a binary executable not fitting the processor architecture. no clue which component or module causes the error but may well be connected with python 3 ? Anybody has an advice on how to debug the script ? BTW: The number "line 138:" and before "illegal instruction" changes randomly and the pairing of the IoT device as 10.42.42.42 does not happen. I have a set of 4 "Nous" smart socket bought last week from amazon. Thank you indeed.

pickup7, 2021-12-17:
Hello back again,
there is definitely a binary executable involved  which does not fit the 32 bit processor of the older Pi.
Flashing Nous socket with a P4 B and finalized without any complaints two minutes ago.
Rgds
pickup7

jakehawkes, 2022-01-03:
This issue is currently preventing me from progressing.
I have a fresh install of Rasbian headless, and a fresh git clone.

plyr0, 2022-01-07:
+1 no success, 138 illegal instruction

Pi Zero W
- raspios 
-- newest official
-- 2021-05-07-raspios-buster-armhf-lite
- raspbian
-- 2020-02-13-raspbian-buster-lite
-- 2019-09-26-raspbian-buster-lite

chriss80, 2022-01-08:
I had the same problem. I solved it by putting the SD card from my Zero W into my Zero 2 W.

malkstar, 2022-01-10:
I received this error today and managed to debug this issue down to the pycryptodomex dependency.

It is repeatable by opening a python3 shell and running `from Cryptodome.Cipher import AES` which results in python crashing with `Illegal operation`, as described in the issue [here](https://github.com/Legrandin/pycryptodome/issues/585). It is apparently an issue with the wheel that the maintainers are looking to get fixed.

The workaround to get `tuya-convert` working for me was to run the following between the `./install_prereq.sh` and `./start_flash.sh` steps in the guide: `sudo -H python3 -m pip install --upgrade pycryptodomex==3.11` 

Hope that helps anyone who comes across this.

plyr0, 2022-01-10:
> The workaround to get `tuya-convert` working for me was to run the following between the `./install_prereq.sh` and `./start_flash.sh` steps in the guide: `sudo -H python3 -m pip install --upgrade pycryptodomex==3.11`

Downgraded the pycryptodomex between `install_prereq.sh` and `start_flash.sh`.
Flashing started without the 138 illegal operation error, which is nice. Newest raspios on rpi zero w.

Flashing still fails with timeout but it might be unrelated, see #1022

Anyway thanks for this workaround @malkstar

rubyrose96, 2022-01-11:
> I received this error today and managed to debug this issue down to the pycryptodomex dependency.
>
> It is repeatable by opening a python3 shell and running `from Cryptodome.Cipher import AES` which [...]

Tried the step you mentioned here and I finally got it to work.
Thanks for working that out!

closed 2022-01-11 by rubyrose96