ct-Open-Source/tuya-convert #1016: Docker vtrust wifi not showing up. No log available

hwr7dd, 2021-12-24:
I'm running Tuya-Convert in a docker on a raspberry pi 4. I get all the way to this step:
sudo docker-compose exec tuya start
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

I cannot find the wifi and I do not see any log in the scripts directory (or the backup directory, it's empty.) 

I'm new to docker so maybe I did not config properly to get logs/make the scripts work?

numbermonkey, 2022-05-28:
Did you ever resolve this? Hitting the same issue and scratching my head.

ZerGo0, 2022-10-09:
If anyone ever has the issue, try restarting your pi and make sure that nothing is running on the ports. I ran the `start_flash.sh` to kill all the services and just killed it after the AP showed up when I was using it on the host. I started the docker container after that and the AP showed up as well. Flashing also worked just fine with the docker setup.