ct-Open-Source/tuya-convert #1019: Unable to flash TYWE3S Device Tower Fan TF4601TR-S

NeilS1966, 2022-01-03:
I'm trying to flash a Stirling Tower Fan purchased from Aldi in Australia. It appears to be the same as the fan listed here: https://templates.blakadder.com/sichler_NX4538-944.html

I opened up the fan to see what micro-controller was installed. It is a TYWE3S.

I'm attempting to flash using a raspberry pi 4.

Tuya convert times out with the following messages:

`Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
No ESP82xx based devices connected according to your wifi log.
Here is a list of all the MAC addresses that connected:
72:16:cc:73:47:29

If you see your IoT device in this list, it is not an ESP82xx based device.
Otherwise, another issue may be preventing it from connecting`

I've previously captured the fan's MAC address and it is not the one listed above.

My log files are provided.
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7804152/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7804153/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7804154/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7804155/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7804156/smarthack-wifi.log)

I'm hoping someone can shed some light on this. I'm keen to get it converted and don't mind resorting to using a soldering iron if necessary.

LloydCapson, 2022-01-06:
Same as https://github.com/ct-Open-Source/tuya-convert/issues/1007

I solved it by running Pi OS Lite - Debian Buster.

NeilS1966, 2022-01-06:
> Same as #1007
> 
> I solved it by running Pi OS Lite - Debian Buster.

Thanks Lloyd, I tried that, same result unfortunately.