ct-Open-Source/tuya-convert #1022: Timed out while waiting for the device to (re)connect

stulluk, 2022-01-07:
Device: NEO COOLCAM WIFI PLUG EU :  http://www.szneo.com/en/products/show.php?id=215
PC: Intel NUC with Ubuntu 21.04 focal (With built-in WIFI adapter)

Here is the failure:

```
ubuntu@ubuntu:~/tuya-convert$ rfkill unblock wifi && ./start_flash.sh 
tuya-convert v2.4.5
Checking for network interface wlp0s20f3... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 4769.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 3s. ..................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] y
======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 3s. ..................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] y
======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..........
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] N
======================================================
Cleaning up...
Closing AP
Exiting...
ubuntu@ubuntu:~/tuya-convert$
```

Logs:
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7830502/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7830503/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7830506/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7830507/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7830509/smarthack-wifi.log)

About 1 year ago, I succesfully converted 3 units ( exactly same model ) with tuya-convert. But now, I can not (tried on brand new two units, both failed)

I allways get this same error message:
**Timed out while waiting for the device to (re)connect**

What I tried:
- Use different computer with a RTL8812BU USB WIFI adapter
- Use a Khadas Edge Arm64 linux SBC with builtin WIFI module
- killall wpa_supplicant
- modify scripts to stop network manager ( systemctl stop NetworkManager for ubuntu ) correctly
- Move WIFI PLUG's to another place / room
- Change WIFI channel to 5 or 6 in hostapd.conf
- Register both samples to original TUYA mobile app, smoke test to make sure they work correctly

Ready to provide more logs if necessary.

What are your suggestions ?

MacSass, 2022-01-09:
Sorry, I don´t have any suggestions for a solution - but the same issue I think.
I have 5 Maxcio plugs, bought at the same time. 2 I could flash with no problem - 3 I´m not able to flash and stuck at:

> SmartConfig complete.
Auto retry in 3s. ..................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
....
> 
The devices go from blinking fast to a state where they either do nothing (no blinking) or they switch the relay about avery 2 minutes ...

For troubleshooting I tried to connect one of them to the official Tuya cloud - which worked right away with no problem (and rendered the device unflashable due to immedeate firmware update - now that one errors out with "firmware is to new...", which is expected).
So now I have 2 plugs left that I´m not able to flash and that just get stuck. I tried to connect a iPhone, an Android phone and even a Windows PC to the vtrust-flash, but that did not make any difference ...

Attached my logs from one try - but I can´t find out from them why the flash does not proceed ... 
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/7835631/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/7835632/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/7835633/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/7835634/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/7835635/smarthack-wifi.log)
 
Any help would be highly appreciated ... thanks.

stulluk, 2022-01-09:
@MacSass I looked at your WIFI and WEB logs, they are pretty similar to mine.

Somehow, device starts getting "upgrade.bin" ( What is that binary by the way? ) and then just starts over and over again. In your case, I see some other weird files, such as "connecttest.txt" and "wpad.dat" ( What are they ? )

I wish there has been a clear sequence diagram in this repository ( **I am volunteer to write & document it** ) to explain how it works, how it mocks tuya's server to intercept upgrade process, what are the contents of this repository, what to do when it fails ( an extensive troubleshooting section )

I hope authors might have a look at our logs, and give us some advices.

MacSass, 2022-01-09:
Hi @stulluk, thanks for looking at those ...
I suspect(!!!) that upgrade.bin might the intermediate firmware that get´s flashed on the devices and then connects to vtrust-flash for the next steps ...
That´s just what I suspect, unfortunately I´m not savy enough to understand the scripts, but would agree that a kind of "flow" of what happens when, with some explanation of the outputs of the steps would be very helpful to better understand what his going on.

Let´s see if one of the authers finds the time to throw in some breadcrumbs for us to help us understand better ...

PS: I will check my logs for "upgrade.bin", the "doing it over and over" might explain why I´m hearing the device flipping the relay periodically about every 1-2 minutes ...

MacSass, 2022-01-10:
Hi, just fyi: I tried the pycryptodomex downgrade mentioned here: https://github.com/ct-Open-Source/tuya-convert/issues/1013#issuecomment-1009337000
just to see if that makes any difference for "our" issue - but for me it does not - it stays exactly the same ...

infocfc, 2022-01-19:
Looks like exactly the same issue current Raspberry images have.
https://github.com/ct-Open-Source/tuya-convert/issues/999#issuecomment-968182311

Which isn't too surprising that you see the same issue on Ubuntu.

kintel, 2022-02-13:
I had exactly the same problem on Ubuntu 21.10 on an Asus chromebook SN60. After finding this ticket, I tried again on an RPi4 running Debian 10.10 and it worked on first try!

stulluk, 2022-02-13:
> I had exactly the same problem on Ubuntu 21.10 on an Asus chromebook SN60. After finding this ticket, I tried again on an RPi4 running Debian 10.10 and it worked on first try!

Thank you so much for your feedback. I currently don't have a ready NEO device at hand, but I will try when I get with my RPi4b.

stulluk, 2022-03-07:
> I had exactly the same problem on Ubuntu 21.10 on an Asus chromebook SN60. After finding this ticket, I tried again on an RPi4 running Debian 10.10 and it worked on first try!

@kintel I get new NEO Wifi PowerPlugs. I have a Rpi4b-4Gbyte model at hand. But I can't find Debian 10.10 image for it. And with newer images ( Debian 11 ), this issue still persists.

Could you please provide the link, where do you download this Debian 10.10 image ?

Or, if you can't find, can you dump your SDcard image with Debian 10.10, and upload to gdrive, so I can download & try ?

kintel, 2022-03-09:
There are some older packages here: https://downloads.raspberrypi.org/raspios_arm64/images/
Not such exactly what versions they are, but some of them should be variations of Debian 10.x.

stulluk, 2022-03-09:
> There are some older packages here: https://downloads.raspberrypi.org/raspios_arm64/images/ Not such exactly what versions they are, but some of them should be variations of Debian 10.x.

I tried all of them, and most images there doesn't boot on my Rpi4b at all. And the ones which are booting, this "timeout" issue persists.

That was the reason , would it be possible for you to dump your SDcard image , so that I can try ?

kintel, 2022-03-09:
I don't have my raw image any longer. My RPi is already fully installed and unfortunately it's not feasible to back out all my private changes.

kintel, 2022-03-10:
Awesome!

tfcarlin, 2022-07-14:
This worked for me as well with a SWB1: [https://www.amazon.com/dp/B07MFTRMM3?psc=1&ref=ppx_yo2ov_dt_b_product_details]

Bulwagga, 2023-01-06:
I had similar issues trying to install from Bullseye and Ubuntu 22.10.  I flashed a minimal Buster environment, installed tuya-convert, and my XS-SSA06 flashed within seconds.  Thank you!

KiralyCraft, 2023-05-16:
Same here, I used `2020-05-27-raspios-buster-armhf.img` on my 3B+ and it worked instantly. Follow stulluk's guide, it worked wonders. On Bullseye it just kept disconnecting the device from the AP, but on buster it worked. Strange, here's to another dump of the SSD :)

zionusxx, 2024-04-26:
Raspberry Pi 4b with Raspberry Pi OS (Bullseye):  

I had an issue with flashing a device because mosquito did not work.
With a configuration file mosquito.conf with those settings it did work:
bind_address 0.0.0.0
allow_anonymous true

Bevor that, it did not bind to the wifi interface and did not allow anonymous login.

https://github.com/ct-Open-Source/tuya-convert/pull/1128

SHU-red, 2025-06-13:
Hi there,
im new to tuya convert and as additional struggle im trying to use it

- on old thinkpad X260
- Running NixOS
- Using a Nix Flake
- Flash target is a already with Tasmota flashed Gosund SP1 (ESP 8266)
- Target binary is a ESPhome already prepared

My Flake seems to run and work

<details>

<summary>flake.nix</summary>

```

{
  description = "Dev shell with Python, MQTT, sslpsk, etc.";

  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-unstable";
    flake-utils.url = "github:numtide/flake-utils";
  };

  outputs = { self, nixpkgs, flake-utils, ... }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          config.allowUnfree = true;
        };

        python = pkgs.python3;

        pythonEnv = python.withPackages (ps: with ps; [
          paho-mqtt
          tornado
          pycryptodomex
          (ps.buildPythonPackage {
            pname = "sslpsk";
            version = "unstable-2021-05-28";
            src = pkgs.fetchFromGitHub {
              owner = "drbild";
              repo = "sslpsk";
              rev = "d88123a75786953f82f5e25d6c43d9d9259acb62";
              sha256 = "sha256-RqaZLtRMzYJPKXBBsw1alujGyqWAQRSQLPyAR8Zi6t4=";
            };
            nativeBuildInputs = [ pkgs.openssl.dev ];
            propagatedBuildInputs = [ pkgs.openssl ];
            pythonImportsCheck = [ "sslpsk" ];
          })
        ]);

      in {
        devShells.default = pkgs.mkShell {
          name = "prerequisites-shell";

          buildInputs = [
            pkgs.git
            pkgs.iw
            pkgs.dnsmasq
            pkgs.util-linux
            pkgs.hostapd
            pkgs.screen
            pkgs.curl
            pkgs.mosquitto
            pkgs.haveged
            pkgs.nettools
            pkgs.openssl.dev
            pkgs.openssl
            pkgs.iproute2
            pkgs.iputils
            pythonEnv
          ];

          shellHook = ''
            echo "✅ Development environment ready."
          '';
        };
      });
}

```

</details>

To properly and (hopefully) be connected with my android phone i perform the following steps

1. `sudo systemctl stop firewall; sudo systemctl stop NetworkManager.service` to be connected permanently, avoiding not getting an ip and let the "You are connected to vtrust-flash" pop up later
2. `nix develop` to open the flake in dev environment
3. `cd ~/Projects/tuya-convert` into the Git-Repository clone on my local pc
4. `./start_flash.sh` to start tuya-convert script
5. Connect with my Android phone (Connected screen pops up)
6. Enter in the Tasmota console of my Tasmota-flashed Gosund SP1 the following command: `WifiConfig 1` to get it to EZ mode? LED slowly blinking Blue\Red ... NOT FAST blinking
7. Then i hit ENTER
8. Tuya convert does not find anything

Can anyone tell me what im doing wrong?

My most uncertain thing is if i properly entered the Pairing/EZ/smarconfig mode of the Tasmota on the Gosund

BTW

I can not find my logfiles .... not sure what i have to do to generate them

<details>
<summary>Terminal window:</summary>

```bash
[shured@nix-laptop:~/Projects/tuya-convert]$ ./start_flash.sh

tuya-convert v2.4.6

Checking for network interface wlp4s0... Found.

Checking UDP port 53... Available.

Checking UDP port 67... Available.

Checking TCP port 80... Available.

Checking TCP port 443... Available.

Checking UDP port 6666... Available.

Checking UDP port 6667... Available.

Checking TCP port 1883... Available.

Checking TCP port 8886... Available.

======================================================

  Starting AP in a screen......

  Starting web server in a screen

  Starting Mosquitto in a screen

  Starting PSK frontend in a screen

  Starting Tuya Discovery in a screen

======================================================

IMPORTANT

1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash

   This step is IMPORTANT otherwise the smartconfig may not work!

2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device

   Make sure nothing else is plugged into your IoT device while attempting to flash.

3. Press ENTER to continue

======================================================

Starting smart config pairing procedure

Waiting for the device to install the intermediate firmware

Put device in EZ config mode (blinking fast)

Sending SSID                  vtrust-flash

Sending wifiPassword

Sending token                 00000000

Sending secret                0101

................

SmartConfig complete.

Resending SmartConfig Packets

.................

SmartConfig complete.

Resending SmartConfig Packets

.................

SmartConfig complete.

Resending SmartConfig Packets

................

SmartConfig complete.

Resending SmartConfig Packets

..................

SmartConfig complete.

Resending SmartConfig Packets

..................
```
</summary>

Any help on this?

I fear i do not have another device/dongle to test and have to get it working with NixOS

Everything feels like its working but somehow the script does not find the device ... maybe im just doing something wrong regarding the EZ mode?