ct-Open-Source/tuya-convert #1023: SSLEOFError when phone connects to wifi

Sarius121, 2022-01-12:
**general information:**
flashed device: CASALUX Wifi LED RGB Light Strip (TYWE3S chip)
operating system: Raspberry Pi 3B+ with Raspberry Pi OS bullseye (headless) and its internal wifi chip

**issue**
I tried to flash the device when I got an timeout:
`Timed out while waiting for the device to (re)connect`

Searching for errors in the logs I found this error in the `smarthack-psk.log`:
```
new client on port 443 from 10.42.42.27:46220
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.27:46232
could not establish sslpsk socket: EOF occurred in violation of protocol (_ssl.c:1123)
Traceback (most recent call last):
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 61, in new_client
    ssl_sock = sslpsk.wrap_socket(s1,
  File "/usr/local/lib/python3.9/dist-packages/sslpsk/sslpsk.py", line 110, in wrap_socket
    sock.do_handshake()
  File "/usr/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:1123)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 112, in <module>
    main()
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 108, in main
    p.data_ready_cb(s)
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 80, in data_ready_cb
    self.new_client(_s)
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 72, in new_client
    if "NO_SHARED_CIPHER" in e.reason or "WRONG_VERSION_NUMBER" in e.reason or "WRONG_SSL_VERSION" in e.reason:
TypeError: argument of type 'NoneType' is not iterable
```
**solution**
This error was caused by my phone and not the device and shouldn't matter for the flashing process. The first error is catched but the second error (throwed because `e.reason` seems to be `None`) not so it prevents the flashing process from working properly for the next connected devices. So I tried just preventing this error by inserting the following after line 71 in the `new_client`-function in psk-frontend.py:
```
[...]
except ssl.SSLError as e:
	print("could not establish sslpsk socket:", e)
	# added following 2 lines:
	if e.reason == None:
		return
	if "NO_SHARED_CIPHER" in e.reason or "WRONG_VERSION_NUMBER" in e.reason or "WRONG_SSL_VERSION" in e.reason:
		print("don't panic this is probably just your phone!")
[...]
```
Because the more detailed error message is not at all important for the flashing process, this change shouldn't effect it.

After this small change everything worked and I successfully flashed my device.

I hope this issue may help others who get the same error.

closed 2022-01-12 by Sarius121

MacSass, 2022-01-12:
Hi @Sarius121,
I´m not as deep into this as you, but I found the same error in my smarthack-psk.log for the issue I did report here:
https://github.com/ct-Open-Source/tuya-convert/issues/1022

Would you mind taking a look and give your opinion if that might be related? For sure I will simply try to add your lines to see if that makes a difference ... I guess it would be interesting to understand why this exception is happening for 3 of my 5 devices (tried all to flash in the same setup).

Thanks!

MacSass, 2022-01-12:
Hi, I think our issue must be a differnent one - as much as I can say from our logs there is no other devices other than the phone ever connecting ... the exception we see in our log at the end seems to come from an interrupt ...

Out of curiosity I tried your modification but it did not make any difference - just the exception not showing up in my log anymore ...

My guess ...

Sarius121, 2022-01-13:
Hi @MacSass,
the error in your smarthack-psk.log seems to be exactly the same and also seems to be triggered by your phone. But if you still get a timeout after adding the two lines and not getting this exact error anymore the problem is probably somewhere else.

> the exception we see in our log at the end seems to come from an interrupt

As far as I understand the program, this interrupt is just the consequence of the timeout and not directly related to an error so your problem is probably not related to an error in the smarthack-psk part of the program.

MacSass, 2022-01-13:
Thanks @Sarius121 - yes I tend to agree with you - some other reason seems to prevent the device from re-connecting with IP 10.42.42.42, but I´m not clear yet on what could be causing this ... I keep searching ...