ct-Open-Source/tuya-convert #1025: Stuck at Starting AP in a screen

andrewprime1, 2022-01-13:
I am somewhat of a noob with regard to the command line, so I am going to just give as much detail as I can as I don't know what is pertinent. Running Tuya Convert on a Raspberry Pi and have successfully flashed several devices with this setup over the last 18 months. Followed [this guide](https://siytek.com/tuya-convert-and-raspberry-pi/) initially for the setup. Just today, before attempting a flash, I ran the get update, get upgrade commands to make sure everything was up to date. Connecting to the Pi via PuTTY

After starting Tuya Convert with sudo ./start_flash.s I am getting stuck at "starting AP in a screen". It just freezes up there and the terminal will not accept any other commands and I have to go unplug my device to reconnect. I tried reading through the issues section here and doing some Googling, but I haven't found any solution. Any tips?

xbavnu, 2022-02-26:
Did you find a solution for this issue? I am having the same problem.

AritroSaha10, 2022-04-08:
Very late response, but the issue (at least for me) was that I was only using wi-fi. Since it starts a hotspot, it ends up disconnecting from the wi-fi network. This would break the SSH tunnel, which lead to it being stuck at that stage. It either needs to be controlled directly (keyboard/mouse) or connected to Ethernet as well.

BJDayzz, 2022-11-24:
I had the same issue today. The problem seems to be the SSH connection. The terminal did freeze. But when I connected the Pi directly to monitor + keyboard it worked.

luckiesdavid, 2023-07-19:
i have the same issue...

dominicdoty, 2023-09-20:
I encountered this issue on my laptop+docker. I resolved it by disconnecting from my wifi network and disabling auto connect on that network.

leolurch, 2024-12-14:
Had the same problem without realizing: I accidentally flashed old pre-configuration through the raspberry pi imager and hence the pi connected to wifi, even though I plugged it in via LAN and never actively configured WIFI...

This thread saved me!