ct-Open-Source/tuya-convert #1026: BK7321t based devices

btsimonh, 2022-01-18:
Hi, I've been looking into a BK7321 based device.

I've successfully flashed it via uart, and feel that alternate firmware can be built.

Do we have a feeling as to if the OTA scheme is the same across multiple processors?
If so, is the source for 'upgrade.bin' available for consideration of generating one for BT7321?
Would I be wasting my time if Tuya have updated the firmwares to close this method of upgrade?

br,

Simon

btsimonh, 2022-01-18:
to add:
The inspiration is from this thread:
https://www.elektroda.com/rtvforum/topic3850712.html#19824097

WolliBolli, 2022-01-18:
in meantime i would Love it.

btsimonh, 2022-01-21:
Ref bk7231, I have published two repos on github:

https://github.com/btsimonh/hid_download_py
https://github.com/btsimonh/tuya-iotos-embeded-sdk-wifi-ble-bk7231t

The 'downloader' can read/write but also unpackage/decrypt original firmware.
The SDK is a 'light touch' modified SDK (including @p.kaczmarek2 mods and samples, although his samples won't work without tuya lib), but removes the tuya lib, and my sample worked (to some degree) with no tuya involved.  Still needs cygwin....
As my devices is now dead (little OTA flash development mistake), my dev stops for a while.  If anyone thinks an STLink can program these chips, pls. point me in the right direction! :)
Br, btsimonh

openshwprojects, 2022-01-29:
BK7231T update, now compatible with Home Assistant and with device configurator, first public release:

https://www.elektroda.com/rtvforum/topic3866123.html

https://github.com/openshwprojects/OpenBK7231T

See Elektroda topic for full documentation and description of current progress and implementation

khalednassar, 2022-02-07:
Hi @openshwprojects and @btsimonh, noticed you've got quite some nice progress on these so far!
FWIW, I've been working with couple of friends for a while on a remote exploit chain for BK7231-based devices to allow local OTA updates for custom firmware. We managed to build an exploit chain for all of those we tried, and will share details when possible.

A bit more details here: https://twitter.com/kmhnassar/status/1490796195886809092

btsimonh, 2022-02-07:
@khalednassar - we have been asking people to keep dumps :).
Looking forward to a connectionless flash!

btsimonh, 2022-02-09:
@khalednassar - what exactly do you need from the device?  - it may be in the bootlog, which is a lot easier to get than the flash read....

khalednassar, 2022-02-11:
@btsimonh at least the app RBL is necessary so there's no way around reading flash

MaxNop, 2022-02-11:
@khalednassar  
Hi I am new to this topic but just managed to read my device with my CH340 like described on elektroda.
What is RBL? What should I do with my readout?
Max

khalednassar, 2022-02-11:
@MaxNop nothing yet besides keeping it handy. will share details soon when we can, so please wait until then

MaxNop, 2022-02-11:
Cool, looking forward to it!
But please tell me what is RBL? Is it the bootloader?

btsimonh, 2022-02-11:
RBL is the format Beken use for firmware - specifically an OTA firmware file is .rbl extension.
But it also is used as a marker for blocks which mark internal partitions.  If you search your flash dump, you will find 'RBL' at least once, if not twice.
A Tuya firmware upgrade file is the .rbl with an additional header....

jagheterfredrik, 2022-02-11:
So a bug in RBL header parsing?

MaxNop, 2022-02-11:
Thank you @btsimonh.  Now as flashing works for me I will try to setup an environment to compile the software myself. I guess I will stumble over RBL then.

btsimonh, 2022-02-11:
@jagheterfredrik  - No, I suspect they need a function address.  Depends how deep the exploit goes, but typically a buffer overflow exploit, for example, would involve placing code in the corrupted memory.  For this code to be useful, it needs addresses of useful functions to call.  Depends on how much code you can inject - maybe we could inject enough code to search for the functions we need? - or dump the flash via wifi?  endless possibilities if you try hard enough :).

axcs, 2022-02-12:
Hello I have several devices with the WB2S Module.
Unfortunately I only noticed it now after having spent the money.
How can I help, will it ever be possible to change the firmware to something like OpenBK7231T by OTA or TUYA-CONVERT?
My knowledge is still a little limited, but if I can help in something so that the WB2S devices can have another firmware without having to open all the devices and solder cables etc.

khalednassar, 2022-04-03:
The tools and exploit for BK7231 devices are finally up on https://github.com/khalednassar/tuya-cloudcutter for the interested.

Usage requires manually making an exploit payload for every distinct firmware app code. This can mean almost every device needs a specialized exploit payload, or we've sometimes found that similar devices from the same brand use the same app code so can share the same exploit payload. But in general the firmware has to be retrieved from at least once per device model so it can be applied on unopened ones of the same exact model.

jagheterfredrik, 2022-04-03:
Thanks for the update and the write-up @khalednassar 👍 nicely done

openshwprojects, 2022-07-24:
OpenBeken now supports BK7231T, BK7231N, BL602, XR809 and W800

Many things has been added during last months, including power metering drivers, RGBCW drivers, Tasmota Devices Groups support and much more:
https://github.com/openshwprojects/OpenBK7231T_App