ct-Open-Source/tuya-convert #1035: could not establish sslpsk socket: EOF occurred in violation of protocol (_ssl.c:1123)

widewing, 2022-02-11:
Hi I'm trying to flash the KissLink SP200. It timeout after sending the config. The led turned solid after this step. In the `smarthack-psk.log` it shows `could not establish sslpsk socket: EOF occurred in violation of protocol (_ssl.c:1123)`

console log:
```
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
....Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
..................
SmartConfig complete.
Auto retry in 1s. ..........................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] n
======================================================
Cleaning up...

```

smarthack-psk.log:
```
new client on port 443 from 10.42.42.40:48382
could not establish sslpsk socket: EOF occurred in violation of protocol (_ssl.c:1123)
Traceback (most recent call last):
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 61, in new_client
    ssl_sock = sslpsk.wrap_socket(s1,
  File "/usr/local/lib/python3.9/dist-packages/sslpsk/sslpsk.py", line 110, in wrap_socket
    sock.do_handshake()
  File "/usr/lib/python3.9/ssl.py", line 1309, in do_handshake
    self._sslobj.do_handshake()
ssl.SSLEOFError: EOF occurred in violation of protocol (_ssl.c:1123)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 113, in <module>
    main()
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 109, in main
    p.data_ready_cb(s)
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 80, in data_ready_cb
    self.new_client(_s)
  File "/home/pi/tuya-convert/scripts/./psk-frontend.py", line 72, in new_client
    if "NO_SHARED_CIPHER" in e.reason or "WRONG_VERSION_NUMBER" in e.reason or "WRONG_SSL_VERSION" in e.reason:
TypeError: argument of type 'NoneType' is not iterable
```

Dalek-mechanic, 2022-08-22:
Hello,
I had the same Error flashing some smart plugs.
Solved it by switching to an older OS (Raspian/Debian 10) and executing the scripts there.

rbm78bln, 2023-04-19:
> it shows could not establish sslpsk socket: EOF occurred in violation of protocol (_ssl.c:1123)

> Solved it by switching to an older OS (Raspian/Debian 10)

Can confirm both. Raspbian Buster worked for me as well a few minutes ago. (rpi3b+)
Running on x64/docker failed for me as well with the exact same error mentioned before.

Target device was BlitzWolf BW-LT21.