ct-Open-Source/tuya-convert #1041: run check for hostapd installed

TRusselo, 2022-03-06:
hostapd is required but not always installed on every system.

run dependency check for it or install with script please.

kueblc, 2022-03-09:
Hi @TRusselo, thanks for your feedback.

`hostapd` is [already installed during set up](https://github.com/ct-Open-Source/tuya-convert/blob/61ba060b8567eacd02813db431967cbe9fd23401/install_prereq.sh#L6). Can you confirm that you've run the `install_prereq.sh` script?

aruhri, 2022-12-22:
Hi,
I had the same issue.
I ran `install_prereq.sh` twice, but had to install `hostapd` manually.

```
lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy
```