ct-Open-Source/tuya-convert #1044: [Answer] Why they fight people wanting to control their own device

qkum, 2022-03-28:
The only reason to why they do everything they can to stop people from re-flashing their own devices is because they sell your data.

That is the only possible answer.

That is how they make the big money - our data that they gain through the app and sell to 3d party like facebook, google, and maybe even governments. 

Hey, they even got full control over the cameras in your house. What 3 letter agency would not drowl and pay big cash for data access to spy on their suspects?

cjj25, 2022-06-29:
While I agree data mining is a huge privacy problem, I believe Tuya is more interested in its profit margin.

You (anyone) can register on Tuya and become a manufacturer, you'll pay a small license fee per device and you can brand those devices as 'QKum Smart Devices Inc.' - However, why would this new QKum business want to pay Tuya license fees when they know all their devices will be flashed with different firmware by the customer? 

Tuya should really offer the consumer a one-off disassociation fee, this would allow the reseller 'QKum Smart Devices Inc' to have any ongoing associated fees removed and/or reimburse 50% of the original license fee.  This would also allow Tuya to continue its profit, while freeing up resources on its servers.

Jojopirat, 2022-11-15:
It's probably a lot more mundane. The manufacturer is printed on the devices. Problems may arise if someone installs firmware other than that authorized by the manufacturer. Customers then often call the manufacturer. This causes considerable effort. That's why the manufacturers want only the software they intended to be used on the devices. Tuya fulfills their wish.
If one day a house burns down and it turns out it was a Tuya device, it will be hard to tell that software not authorized by the manufacturer was running on it.