ct-Open-Source/tuya-convert #1046: ACTION "LSC Smart Connect Einbauspot" won't flash

plin2, 2022-04-19:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
no

**Do you have a question about configuring your newly flashed device?**
no

**Did you have issues getting a device to flash?**
yes

**Please remember to close your issue when the problem has been addressed!**

I tried to flash this Downlight (https://www.action.com/de-de/p/lsc-smart-connect-einbauspot/), didn't succeed

My log shows
`./start_flash.sh
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
..............
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
Your device's firmware is too new.
Tuya patched the PSK vulnerability that we use to establish a connection.
You might still be able to flash this device over serial.
For more information and to follow progress on solving this issue see:
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02
======================================================
Do you want to try flashing another device? [y/N] y
======================================================
`

Does anybody else have expecirence with this device?

imkebe, 2022-04-22:
You've got all information printed. Had you read it? It won't work.

plin2, 2022-04-24:
Well, I saw

```
Your device's firmware is too new.
Tuya patched the PSK vulnerability that we use to establish a connection.
You might still be able to flash this device over serial.
For more information and to follow progress on solving this issue see:
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02
```

that's why I'm asking for 'expericne' with this device.

Is the chip built into the decive suitable for Tasmota?
Is the PSK key the only problem?
...

HumbleDeer, 2022-08-27:
Hi,

Can you give me physical details or photos of the details of this device? I can pick one up at Action or have a look for you whether this would be easily flashable through UART.