ct-Open-Source/tuya-convert #1050: ESP_1AE1A6 (eufy robovac) not discovered

fundef1, 2022-05-26:
in short: can't flash the device - is it not supported, or is something else wrong?

I have a (slightly) different setup so logs aren't that helpful, but I have collected to info that I could.
My setup: 
* A Eufy robovac 30C - never connected to the internet, unclear what version it has or how to check
* separate Access point: with vtrust-flash SSID, running a local dnsmasq
* laptop with debian11 running the tuya-convert scripts with wifi/ap/dhcp bits commented out in setup_ap.sh 
* * with wired connection to the AP, so WIFI in config.text is enp2s0
* "any other device" connected to the vtrust-flash ssid

after running apconfig.py the situation is as follows, all devices connected to vtrust-flash:
* robovac: hostname ESP_1AE1A6	mac: C8:2B:96:1A:E1:A6 ip: 10.42.42.22
* * port 6668 seems to be open but nothing else
* * 1AE1A6 is not in the OUI list of ESP/non-ESP but the hostname is confusing...
* AP: 10.42.42.2
* debian: ip 10.42.42.1

## the device is not in EZ config mode 
(not sure what that is, or how to trigger it - but it shouldn't be required - if I understand apconfig.py correctly)

Smartconfig (which sends out a broadcast many time ?) times out, logs essentially empty:
* tuya-discovery (-udp.log ) doesn't list anything except listening on 6666/6667
* mosquito (-mqtt.log) listening on 1883
* web listening on port 80
* -psk is empty
* (-wifi obviously not relevant)

(a version of ) the robovac is flashable as listed on the tasmota page, but maybe mine is not.... or somebody can find a mistake in my setup? any help would be much appreciated.

Thanks!

kueblc, 2022-05-28:
Hi @fundef1,

Sounds like you're most of the way there. You've got the device configured for the `vtrust-flash` network, so the next step is getting the device to talk to our mock server.

Is your dnsmasq configured to send all traffic to your machine running the fake server script?

Can you check the server script output for any signs of connection?

fundef1, 2022-05-29:
hi @kueblc thank for the follow-up... yeah, i think/hope i'm close, just not there yet.
in short: no dnsmasq wasn't specifically configured at all. I didn't see any dnsmasq config in the scripts either. if you can let me know which dns entries to point to where i'll give that a shot.

anyway, in the mean time I've simplified the setup a bit. I now have an AP running on the debian machine. and the scripts run as-is. (had to add a sudo on the AP capability check with iw) but now there's a bit more life. the mosquitto logs look suspicious, i'll dig a bit deeper... in the meantime here are the logs as right now::

```
1653811217: mosquitto version 2.0.11 starting
1653811217: Using default config.
1653811217: Starting in local only mode. Connections will only be possible from clients running on this machine.
1653811217: Create a configuration file which defines a listener to allow remote access.
1653811217: For more details see https://mosquitto.org/documentation/authentication-methods/
1653811217: Opening ipv4 listen socket on port 1883.
1653811217: Opening ipv6 listen socket on port 1883.
1653811217: mosquitto version 2.0.11 running
^C1653811365: mosquitto version 2.0.11 terminating
```
```
new client on port 443 from 10.42.42.13:42364
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.13:42374
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
^CTraceback (most recent call last):
  File "/home/jo/tuya-convert/scripts/./psk-frontend.py", line 113, in <module>
    main()
  File "/home/jo/tuya-convert/scripts/./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt
```
```
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C
```
Listening on 10.42.42.1:80
[32m[I 220529 10:00:36 web:2239][m 302 GET /generate_204 (10.42.42.13) 1.87ms
[32m[I 220529 10:00:36 web:2239][m 200 GET / (10.42.42.13) 1.21ms
[32m[I 220529 10:00:36 web:2239][m 302 GET /generate_204 (10.42.42.13) 0.74ms
[32m[I 220529 10:00:37 web:2239][m 302 GET /generate_204 (10.42.42.13) 0.66ms
[32m[I 220529 10:00:37 web:2239][m 304 GET / (10.42.42.13) 1.01ms
[32m[I 220529 10:00:38 web:2239][m 302 GET /favicon.ico (10.42.42.13) 0.66ms
[32m[I 220529 10:00:38 web:2239][m 304 GET / (10.42.42.13) 0.73ms
[32m[I 220529 10:00:38 web:2239][m 302 GET /generate_204 (10.42.42.13) 0.62ms
[32m[I 220529 10:00:40 web:2239][m 302 GET /generate_204 (10.42.42.13) 0.94ms
^CReceived SIGINT, exiting...
```
System info
===========
61ba060
Linux tosh 5.10.0-13-amd64 #1 SMP Debian 5.10.106-1 (2022-03-17) x86_64 GNU/Linux
OpenSSL 1.1.1n  15 Mar 2022
Dnsmasq version 2.85  Copyright (c) 2000-2021 Simon Kelley
Compile time options: IPv6 GNU-getopt DBus no-UBus i18n IDN2 DHCP DHCPv6 no-Lua TFTP conntrack ipset auth cryptohash DNSSEC loop-detect inotify dumpfile

This software comes with ABSOLUTELY NO WARRANTY.
Dnsmasq is free software, and you are welcome to redistribute it
under the terms of the GNU General Public License, version 2 or 3.
hostapd v2.9
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
Python 3.9.2
===========
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Failed to stop network-manager.service: Unit network-manager.service not loaded.
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp3s0...
Configuration file: /dev/stdin
Using interface wlp3s0 with hwaddr 1a:6a:65:df:04:de and ssid "vtrust-flash"
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: AP-STA-CONNECTED c8:2b:96:1a:e1:a6
wlp3s0: AP-STA-CONNECTED 60:ab:67:e8:0f:95
wlp3s0: interface state ENABLED->DISABLED
wlp3s0: AP-STA-DISCONNECTED 60:ab:67:e8:0f:95
wlp3s0: AP-STA-DISCONNECTED c8:2b:96:1a:e1:a6
wlp3s0: AP-DISABLED 
wlp3s0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlp3s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...
Failed to restart network-manager.service: Unit network-manager.service not found.
```

fundef1, 2022-05-29:
update, added a default mosquitto.conf with a single listener on 1883. Behaviour is still the same, no connections logged.
(I believe i previously didn't see the mosquitto conf issue as i was running everything as root, now just as a regular user)

```
1653812866: mosquitto version 2.0.11 starting
1653812866: Config loaded from /home/jo/tuya-convert/scripts/mosquitto.conf.
1653812866: Opening ipv4 listen socket on port 1883.
1653812866: Opening ipv6 listen socket on port 1883.
1653812866: mosquitto version 2.0.11 running
^C1653812995: mosquitto version 2.0.11 terminating
```

kueblc, 2022-05-29:
You'll want to set up your `dnsmasq` to redirect all traffic to the machine running the fake server. Once you get this set up, I think you should be good to go.

https://github.com/ct-Open-Source/tuya-convert/blob/61ba060b8567eacd02813db431967cbe9fd23401/scripts/setup_ap.sh#L39-L47

Note the `--address` argument, where it maps `#` (wildcard) to `$GATEWAY` (by default 10.42.42.1).

>     1653811217: Starting in local only mode. Connections will only be possible from clients running on this machine.

This does seem suspicious, though you've already solved this, I hadn't realized that mosquitto's default behavior had changed. I'll probably want to include a default config in the next release.

I see in your web log that your IoT device is indeed connecting to your `vtrust-flash` AP, so you can skip the pairing step from now on. There is no need to connect a secondary device to the `vtrust-flash`, this is only to facilitate EZ config (and you used AP config anyway).

fundef1, 2022-05-31:
thanks for the response, i did spot the --address line later on, even with that added... still no luck.
Once the ESP_ is connected I can't get any response out of it. 

Am I correct in assuming that the smartconfig.py script is responsible for triggering a firmware update?
there is a region setting US.... any point in fiddling with that? (tried EU but behaviour isn't any different)

kueblc, 2022-06-01:
> alright that does help narrowing it down. so it's essentially a captive portal setup?

Yes you can think of it like a captive portal. We direct all traffic to a fake registration server that allows it to complete the initialization process, trigger the firmware upgrade, and then provide the firmware binary.

> (at least the one on port 80, **not clear on what PSK does yet**)

Older firmware used http in plaintext. Tuya attempted to patch against tuya-convert by implementing a PSK based https. We subsequently found a flaw in their implementation that allowed us to decrypt the traffic and forward it to our fake registration server to keep tuya-convert working. (They have now patched this vulnerability too so tuya-convert only works on first and second gen Tuya devices, continuing the game of cat and mouse)

> so I am _not_ running any of the scripts, only an access point with embedded dnsmasq (including an address=/#/... line)

Are you sure you're mapping # to the IP of the machine running the fake registration server?

> the ESP_ does **not** do any DNS request upon **re**connecting to the AP

This makes sense, the device only needs to resolve the domain once, so if it connected without the captive portal it already had the IP address of the real registration server.

Those requests to resolve `a3.tuyaus.com` are a very good sign indeed.

> so what would the next step in the process be? a request to port ... the pskfrontend.py script?

It should then connect to the fake registration server directly over port 80 (if running 1st gen firmware), or tunneling via PSK frontend over 443 to the fake registration server (if 2nd gen). Fingers crossed it doesn't have patched firmware.

fundef1, 2022-06-04:
Allright! As in literally All is Right ;-)
after putting the webserver back in place it was smooth sailing all the way. 

Thanks for your help.

closed 2022-06-04 by fundef1