ct-Open-Source/tuya-convert #1057: Cannot flash Topgreener TGWF115APM

ipher, 2022-07-31:
Topgreener TGWF115APM, device is printed with "1810Ver1.0.1" on it. Using a Raspberry Pi 3B V1.2 to flash

There seems to be a fault in smarthack-psk.log 

[Console.txt](https://github.com/ct-Open-Source/tuya-convert/files/9229532/Console.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/9229533/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/9229534/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/9229535/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/9229536/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/9229537/smarthack-wifi.log)

Lazymin3r, 2022-08-09:
I am experiencing the exact same issue attempting to flash TopGreener Smart Plug: TGWF115APM. I also have the same version 1810 Ver 1.0.1 printed on the device. I am using a Rasberry Pi 4 running the tuya converter using docker. But seeing the same issue with smartconfig procedure looping with no success.

ebrien, 2022-08-23:
Same here though I have a 1906Ver1.0.1 but the wifi chip has the same markings.
[console.txt](https://github.com/ct-Open-Source/tuya-convert/files/9410969/console.txt)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/9410970/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/9410971/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/9410972/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/9410973/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/9410974/smarthack-wifi.log)

ipher, 2022-08-27:
I tried an unused one that I bought a year or so ago, that had never been connected to the app, and it didn't work either. Since it was never connected to wifi or the app I don't think it was a recent update that prevents this

MacSass, 2022-08-27:
Hi,
I´m not sure this is the same issue I had - but in my case I really tried everything I could think of with no success.

In the end this modified procedure allowed me to flash my devices:
https://github.com/ct-Open-Source/tuya-convert/pull/966#issuecomment-1213645782

Regards - MacSass

ebrien, 2022-08-29:
I'm able to connect to the vtrust-flash SSID from the 2nd computer (Android tablet) and I can see that my TopGreener device also connects to it.
If I follow the procedure and put the device in Pairing Mode after the SSID is available, Tuya-Convert complete the "SmartConfig" and then time out.

```
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.

### Removed the lines for simpliciy ###

3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 3s. ..................................................................................................................
Timed out while waiting for the device to (re)connect

```
When I put the device in pairing mode before starting tuya-convert here is what it happens:

```
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Available.

### Removed the lines for simpliciy ###

3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
....................................
Timed out while waiting for the device to (re)connect

```

MacSass, 2022-08-29:
I had exactly the same issue for 3 of my 5 devices ...
With the apconfig.py script listed on the comment linked above and a manual connect to the device in AP mode it worked perfectly. Somehow forcing the device to reconnect to the AP seems flawed and unreliable.
Using the procedure listed above make the process way more robust, obviously.

floxboy, 2022-09-05:
where can i get appconfig.py? I don't see it in tuya-convert folders

MacSass, 2022-09-05:
Hi,
appconfig.py is not yet part of the official tuya-convert.
If you check the comment mentioned above: https://github.com/ct-Open-Source/tuya-convert/pull/966#issuecomment-1213645782 you can find it in the commits:
https://github.com/ct-Open-Source/tuya-convert/pull/966/commits/9d1d35bb6c195fb82ce96c287696bb85a00467d6

Good luck, it worked perfectly for my devices that I had not been able to flash for weeks trying everything I could think of!

jwidess, 2025-04-23:
> Hi, appconfig.py is not yet part of the official tuya-convert. If you check the comment mentioned above: [#966 (comment)](https://github.com/ct-Open-Source/tuya-convert/pull/966#issuecomment-1213645782) you can find it in the commits: [9d1d35b](https://github.com/ct-Open-Source/tuya-convert/commit/9d1d35bb6c195fb82ce96c287696bb85a00467d6)
> 
> Good luck, it worked perfectly for my devices that I had not been able to flash for weeks trying everything I could think of!

I'm also trying to flash a TGWF115APM without any success. I've tried using the [`apconfig.py`](https://github.com/ct-Open-Source/tuya-convert/commit/9d1d35bb6c195fb82ce96c287696bb85a00467d6) script after putting the device in AP mode and then connecting to the AP on my Linux laptop, which outputs to the terminal `Sending <long string>` then I tried to run `start_flash.sh` again, but I'm getting the same `Timed out while waiting for the device to (re)connect`. If anyone has any ideas, it would be much appreciated!