ct-Open-Source/tuya-convert #1058: Endless loop: Resending Smartconfig Packets

SteveRMann, 2022-08-07:
I am in an endless loop:

```
SmartConfig Complete.
Resending Smartconfig packets
...............................................

```

Eventually timing out.
The MAC address shown in the "connected" list is not the MAC address of the device.

The device is a Geenie 20405 Four outlet + USB smart strip.
**log files**:

[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/9277238/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/9277239/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/9277240/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/9277241/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/9277242/smarthack-psk.log)

Does this device have the firmware that is preventing a new flash or did I miss something in the process?

Smitty357, 2022-11-05:
I am having the same issue with some older 'merkury' bulbs.
The device never shows up in the list of connected devices, but.....
While in rapid flashing mode, the light will stop flashing shortly after it attempts to convert it. This tells me that it is attempting to do something. 
I can let it flash for 10 minutes, then start the process, and shortly after it begins it will stop flashing.
The bulbs are part of a batch I bought 2 years ago. Half of them converted just fine, but never opened these until now.
Anyone have suggestions?

dgagnon720, 2023-05-25:
Same here.  I'm using tuya-convert in docker.  Are you guys running it also in docker?