ct-Open-Source/tuya-convert #1064: Failed to flash WIFIDS10WT

MrMEEE, 2022-09-10:
### Before opening a new issue, please read the following

Trying to flash my Nedis WIFIDS10MT Smoke detector:

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword          
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Auto retry in 1s. ..............................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
Do you want to try flashing another device? [y/N] N
======================================================
Cleaning up...
Closing AP
Exiting...

Can anyone help me out?

MrMEEE, 2022-09-10:
==> smarthack-mqtt.log <==
1662846375: mosquitto version 2.0.11 running
^C1662846470: mosquitto version 2.0.11 terminating
1662846584: mosquitto version 2.0.11 starting
1662846584: Using default config.
1662846584: Starting in local only mode. Connections will only be possible from clients running on this machine.
1662846584: Create a configuration file which defines a listener to allow remote access.
1662846584: For more details see https://mosquitto.org/documentation/authentication-methods/
1662846584: Opening ipv4 listen socket on port 1883.
1662846584: Opening ipv6 listen socket on port 1883.
1662846584: mosquitto version 2.0.11 running

==> smarthack-psk.log <==
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.28:41223
could not establish sslpsk socket: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:997)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.28:41225
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:997)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.28:41227
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:997)
don't panic this is probably just your phone!

==> smarthack-udp.log <==
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C/home/mj/Hentet/tuya-convert/scripts/./tuya-discovery.py:50: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667
^C/home/mj/Hentet/tuya-convert/scripts/./tuya-discovery.py:50: DeprecationWarning: There is no current event loop
  loop = asyncio.get_event_loop()
Listening for Tuya broadcast on UDP 6666
Listening for encrypted Tuya broadcast on UDP 6667

==> smarthack-web.log <==
^CReceived SIGINT, exiting...
Listening on 10.42.42.1:80
[I 220910 23:49:52 web:2271] 302 GET /generate_204 (10.42.42.28) 0.56ms
[I 220910 23:49:52 web:2271] 302 GET /generate_204 (10.42.42.28) 0.70ms
[I 220910 23:49:52 web:2271] 200 GET / (10.42.42.28) 1.04ms
[I 220910 23:49:52 web:2271] 302 GET /favicon.ico (10.42.42.28) 0.67ms
[I 220910 23:49:53 web:2271] 304 GET / (10.42.42.28) 0.70ms
[I 220910 23:49:54 web:2271] 302 GET /generate_204 (10.42.42.28) 0.54ms
[I 220910 23:49:54 web:2271] 302 GET /generate_204 (10.42.42.28) 0.57ms
[I 220910 23:50:12 web:2271] 302 GET /generate_204 (10.42.42.28) 0.66ms

==> smarthack-wifi.log <==
Attempting to stop wpa_supplicant
Stopping NetworkManager...
Configuring AP interface...
RTNETLINK answers: File exists
Starting DNSMASQ server...
Starting AP on wlp3s0...
wlp3s0: interface state UNINITIALIZED->ENABLED
wlp3s0: AP-ENABLED 
wlp3s0: AP-STA-CONNECTED 5c:77:76:97:0a:10
wlp3s0: AP-STA-CONNECTED f4:cf:a2:5c:e2:fb

==> smarthack-web.log <==

GET /gw.json?a=s.gw.token.get&et=1&gwId=04876840f4cfa25ce2fb&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":28,"source":"ap","path":"multicast"}}&t=4&v=3.0&sign=965ffaf3317182d0b9c8391f88395db7
Host: a3.tuyaus.com
User-Agent: ESP8266SDK
Connection: keep-alive

Answer s.gw.token.get
reply {"t":1662846629,"e":false,"success":true,"result":{"gwApiUrl":"http://10.42.42.1/gw.json","stdTimeZone":"-05:00","mqttRanges":"","timeZone":"-05:00","httpsPSKUrl":"https://10.42.42.1/gw.json","mediaMqttUrl":"10.42.42.1","gwMqttUrl":"10.42.42.1","dstIntervals":[],"mqttsUrl":"10.42.42.1","mqttsPSKUrl":"10.42.42.1","mediaMqttsUrl":"10.42.42.1","aispeech":"10.42.42.1"}}
[I 220910 23:50:29 web:2271] 200 GET /gw.json?a=s.gw.token.get&et=1&gwId=04876840f4cfa25ce2fb&other={"token":"00000000","region":"US","tlinkStat":{"configure":"smartconfig","time":28,"source":"ap","path":"multicast"}}&t=4&v=3.0&sign=965ffaf3317182d0b9c8391f88395db7 (10.42.42.1) 25.56ms
[... 228 lines cut ...]
^CTraceback (most recent call last):
  File "/home/mj/Hentet/tuya-convert/scripts/./psk-frontend.py", line 113, in <module>
    main()
  File "/home/mj/Hentet/tuya-convert/scripts/./psk-frontend.py", line 106, in main
    r,_,_ =  select.select(readables, [], [])
KeyboardInterrupt

==> smarthack-web.log <==
^CReceived SIGINT, exiting...

==> smarthack-mqtt.log <==
^C1662846721: mosquitto version 2.0.11 terminating

==> smarthack-udp.log <==
^C
==> smarthack-wifi.log <==
wlp3s0: interface state ENABLED->DISABLED
wlp3s0: AP-STA-DISCONNECTED f4:cf:a2:5c:e2:fb
wlp3s0: AP-STA-DISCONNECTED 5c:77:76:97:0a:10
wlp3s0: AP-DISABLED 
wlp3s0: CTRL-EVENT-TERMINATING 
nl80211: deinit ifname=wlp3s0 disabled_11b_rates=0
AP closed
Stopping DNSMASQ server...
Restarting NetworkManager...

kueblc, 2022-09-13:
Thanks for your detailed logs @MrMEEE 

It looks like this device should be supported, however it is not accepting the fake config.get API response.

I've seen this come up before and at the moment, I do not know of a workaround. I can take another look soon and let you know if I find anything you can try.

west89, 2022-09-27:
> Thanks for your detailed logs @MrMEEE
>
> It looks like this device should be supported, however it is not accepting the fake config.get API response.
>
> I've seen this come up before and at the moment, [...]

The endless loop also occurs with the Nous A1 product, which is identical in construction to the Gosund SP111 and the BlitzWolf SHP6: 

SmartConfig complete.
Resending SmartConfig Packets

olliemaller, 2024-11-09:
i have the same issue. Does anyone have a fix