ct-Open-Source/tuya-convert #1065: Gosund like GSP11

xsasx, 2022-09-27:
Hi all,

started a new raspian os install completly fresh. Got 4 Gosund like SP11 with Model name GSP11 sadly i cant flash them:

could not establish sslpsk socket: [SSL: WRONG_SSL_VERSION] wrong ssl version (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.25:64079
ID: 02480901df2734a73f0957b3dc664f1266126d3ebf8d7ef6756cecf4f7803122df249301d45324da9c3bf14d03daf9c87b
Prefix: b"H\t\x01\xdf'4\xa7?\tW\xb3\xdcfO\x12f"
PSK: 2f396b0df97be9e93147f59e440914e9dea53fcfbda21dbeae7f82b1b1c5b16c
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.25:64080
ID: 029bb159e7296810a660b8ca26ba5a46f5126d3ebf8d7ef6756cecf4f7803122df249301d45324da9c3bf14d03daf9c87b
Prefix: b'\x9b\xb1Y\xe7)h\x10\xa6`\xb8\xca&\xbaZF\xf5'
PSK: 2d17a5619ef801e479273b30f6ddbd41874e11b99e5893d827c8171fc7b71f9d
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.25:64081
ID: 0231ceec2ba9642f722b3b6eb4839a5922126d3ebf8d7ef6756cecf4f7803122df249301d45324da9c3bf14d03daf9c87b
Prefix: b'1\xce\xec+\xa9d/r+;n\xb4\x83\x9aY"'
PSK: 8c4f48aaad34a1bf80193cd3d0ecc8bcbce6aecae0e7fe1372ca3949bd57dbcb
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.25:64082
ID: 0299a376142151fc02e275f11f9b6652dc126d3ebf8d7ef6756cecf4f7803122df249301d45324da9c3bf14d03daf9c87b
Prefix: b'\x99\xa3v\x14!Q\xfc\x02\xe2u\xf1\x1f\x9bfR\xdc'
PSK: ef15f6f6572e865140e5e5c201513a4468c637a94e8d88b40855b0d354d9e0d4
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.25:64083
ID: 029cda917a084425e268be5e60be85d49f126d3ebf8d7ef6756cecf4f7803122df249301d45324da9c3bf14d03daf9c87b
Prefix: b'\x9c\xda\x91z\x08D%\xe2h\xbe^`\xbe\x85\xd4\x9f'
PSK: cbb52e26f8f8aa8e48dcc00e0abc02110445dbbed1a0abae61002e83d90e444e
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.22:62211
ID: 021f7dd068da91d249b3d060c5f5202b2f998477474826822e4533504dc0ecf11a3c11b505656a0a1f49bf128d6cf5ced3
Prefix: b'\x1f}\xd0h\xda\x91\xd2I\xb3\xd0`\xc5\xf5 +/'
PSK: cfd9f300c18887110da434b6ed5fb5deeeb8308662b5b94f039032532e4b0305
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.22:62212
ID: 0289a11091d97389a646f1cf492f4942a1998477474826822e4533504dc0ecf11a3c11b505656a0a1f49bf128d6cf5ced3
Prefix: b'\x89\xa1\x10\x91\xd9s\x89\xa6F\xf1\xcfI/IB\xa1'
PSK: 756655e026bae66f1aa0ed2610794697575d21f9dac9dda5584004711afd9506
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.22:62213
ID: 0231a81393fb34e8c1b0046c0a04ed46bd998477474826822e4533504dc0ecf11a3c11b505656a0a1f49bf128d6cf5ced3
Prefix: b'1\xa8\x13\x93\xfb4\xe8\xc1\xb0\x04l\n\x04\xedF\xbd'
PSK: b3da0869f2520a2d8e681b4bdad67a1da9c64bf2a9201dbce3fb17c05177f4d0
could not establish sslpsk socket: [SSL: DECRYPTION_FAILED_OR_BAD_RECORD_MAC] decryption failed or bad record mac (_ssl.c:1123)
new client on port 443 from 10.42.42.20:59035
could not establish sslpsk socket: [SSL: NO_SHARED_CIPHER] no shared cipher (_ssl.c:1123)
don't panic this is probably just your phone!
new client on port 443 from 10.42.42.20:49208
could not establish sslpsk socket: [SSL: WRONG_VERSION_NUMBER] wrong version number (_ssl.c:1123)

Can anybodey helmp me out ?

TrustedFloppa, 2022-12-17:
Try Connecting vtrust-flash to a laptop or smt else
[image]

gnagy, 2023-02-18:
I was able to get it working on Debian 11 with downgrading openssl security level. At the end of `/etc/ssl/openssl.cnf` I changed the section to this:

```
[system_default_sect]
MinProtocol = TLSv1.1
CipherString = DEFAULT@SECLEVEL=0
```