ct-Open-Source/tuya-convert #1066: C168/188 flashed, connected to wifi, now doesn't turn on

mholl1983, 2022-10-02:
Hi, I flashed the C168/188 smart outdoor plug. I connected to 192.168.4.1 and set up the WiFi/credentials (5G network). After the device restarted, it didn't come online and there are no lights on the device. I tried to do a reset on the device but that didn't work either; still no lights.

I believe the device is bricked. Wondering if it prefers a 2.4 ghz network. Either way, is there a way I can safely do a factory reset or reflash the device? Thanks in advance!

MacSass, 2022-10-02:
I have never seen any device working on 5 GHz network - all the current ESPs from espressif work on 2.4 GHz only.

mholl1983, 2022-10-02:
Understood. I'd like to redo the initial config and choose the 2.4 ghz option, but I don't know how to start from scratch when the device won't turn on nor go into pairing mode.

kueblc, 2022-10-02:
It's likely trying to connect to the network you provided credentials for. You can set up a 2.4GHz network with those same credentials temporarily, so that you can connect to it and reconfigure.

As @MacSass said, the ESP82xx only supports 2.4GHz. It may be that your network uses the same SSID for 2.4 and 5. If you truly had it connected to a 5GHz network, there's another unsupported microcontroller in your device. But if that were the case, then neither tuya-convert nor Tasmota should have succeeded.

mholl1983, 2022-10-02:
Thanks for the reply! That's an interesting idea about doing the 2.4 ghz network as a temporary measure to try to reconfigure. I'll give that a try sometime soon, thanks,

Yeah, I turn off band steering on my network, always. I don't trust it, so we have separate streams altogether for 2.4 and 5.

MacSass, 2022-10-02:
If your device has a button, you should be able to reset device config by holding it an extended time (think 40 secs). Another method is to power on/off  6 times in a row ...
Check out the guides - your case is mentioned there ... I think it also depends a bit on Tasmota version:
https://tasmota.github.io/docs/Device-Recovery/
https://tasmota.github.io/docs/FAQ/

mholl1983, 2022-10-03:
@MacSass , thanks very much! Thanks to your guidance/link above, I was able to do a reset on the device (power on/off method is what worked in the end). I saw the device w/ its temporary AP show up right away and was able to reconfigure it. It's not connected to 2.4 ghz and working just fine. Many thanks for your help, again! My parents have the same device, so I'm looking forward to "declouding it" for them, too. Cheers!

closed 2022-10-03 by mholl1983