ct-Open-Source/tuya-convert #1070: CE smart Home LA-WF3 Unable to flash

Bun-Bun, 2022-10-16:
Trying to flash a CE smart Home LA-WF3. Bought it years ago in a two pack. Never used it. The other one in the pack I dismantled and flashed via wire tasmota. Thought I would save myself the hassle and try tuya-convert! No go. 

I looked through the logs, nothing stands out to me. 

`======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 1s. ...............................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from flashing.
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.
======================================================
`
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/9793503/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/9793504/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/9793505/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/9793506/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/9793507/smarthack-psk.log)

Bun-Bun, 2022-10-16:
I started over and flashed Raspberry Pi OS Buster Lite (legacy). 

Worked as expected. Flashed both my devices without issue. 

Something in the newer image, or the face I used 64bit, was messing something up.

davelosert, 2022-12-23:
Thank you, that did the trick for me!
Also tried it with the Standard Buster Lite with the same issue - using the legacy one, it now works!