ct-Open-Source/tuya-convert #1072: Tuya Convert does not work with latest Raspberry Pi Bullseye OS

schneemaier, 2022-11-24:
Tried with multiple devices under latest Raspberry Pi OS (Bullseye) and flashing did not work. Using the same procedure Buster (previous version) ended with successful flash on the same devices.

Oneboy1979, 2022-11-28:
Works like charm with Raspberry PI 3b+ and bullseye.

schneemaier, 2022-11-28:
@Oneboy1979 Are you using a fresh install of bullseye or an upgrade from previous version? I tried it on a vanilla bullseye (latest Raspberry Pi OS Lite Release date: September 22nd 2022) also on a Raspberry PI 3b

Oneboy1979, 2022-11-28:
@schneemaier i use a fresh install and as i wrote no problem for me

schneemaier, 2022-11-29:
Thanks. I don't have time now to test it again, but will do it. I will report back my fidings

evertos, 2023-02-20:
Well did have this issue as well! I did flash all my devices. But if get a new one I will make sure to try it again!
One thing to note, I did try it from a PI that is used as an Octoprint server. If you want the logs from this Please let me know!