ct-Open-Source/tuya-convert #1073: Gosund Sp1 flash hat nicht funktioniert (vtrust-recovery)

cncoding99, 2022-11-28:
Ich habe meine Gosund Sp1 flashen wollen. Erstes Gerät hat funktioniert, bei den anderen (selbes Kaufdatum) hat es nicht funktionier:

`pi@raspberrypi:~ $ cd tuya-convert
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.5
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 591.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 522.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen...
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fa                          st). This is usually done by pressing and holding the primary button of the devi                          ce
   Make sure nothing else is plugged into your IoT device while attempting to fl                          ash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware

IoT-device is online with ip 10.42.42.42
Stopping smart config
Fetching firmware backup
mkdir: cannot create directory ‘../backups/20221128_180714’: Permission denied
./start_flash.sh: line 105: pushd: ../backups/20221128_180714: No such file or directory
======================================================
Cleaning up...
Closing AP
Exiting...
`

Kann mir jemand helfen?

#265 ähnliches Problem, aber ich verstehe die Lösung nicht...

kueblc, 2022-11-28:
> mkdir: cannot create directory ‘../backups/20221128_180714’: Permission denied
> ./start_flash.sh: line 105: pushd: ../backups/20221128_180714: No such file or directory

Looks like the user running the script does not own the `backups` directory. Fix your permissions then you should be able to run the script again. Since you already paired the IoT device, you can skip this step (press enter) and then the backup should proceed as normal.

kueblc, 2022-11-28:
Using your standard unix commands, eg,
```
cd tuya-convert
sudo chown -R backups
```

cncoding99, 2022-11-28:
@kueblc which operand do I have to use? Sry I have never done this bevor

`pi@raspberrypi:~/tuya-convert $ sudo chown -R backups
chown: missing operand after ‘backups’
Try 'chown --help' for more information.
pi@raspberrypi:~/tuya-convert $
`

kueblc, 2022-11-28:
Since you are running as user `pi`, it would be
```
sudo chown -R pi backups
```

cncoding99, 2022-11-28:
Thanks it worked :)

closed 2022-11-28 by cncoding99