ct-Open-Source/tuya-convert #1075: Unable to pull tuya:latest. No longer exists at docker-hub.

aaronouthier, 2022-12-09:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

stnokott, 2023-01-04:
You need to build it first using the `Dockerfile` in the repo.

`docker build -t tuya:latest .`

The Docker documentation in the README mentions this step.

sewo80, 2023-08-14:
its not working anymore:
# docker build -t tuya:latest .
[+] Building 2.2s (3/3) FINISHED                                                                                                      docker:default
 => [internal] load build definition from Dockerfile                                                                                            0.1s
 => => transferring dockerfile: 426B                                                                                                            0.0s
 => [internal] load .dockerignore                                                                                                               0.1s
 => => transferring context: 2B                                                                                                                 0.0s
 => ERROR [internal] load metadata for docker.io/phusion/baseimage:master                                                                       0.4s
------
 > [internal] load metadata for docker.io/phusion/baseimage:master:
------
Dockerfile:1
--------------------
   1 | >>> FROM phusion/baseimage:master
   2 |     
   3 |     RUN apt-get update && apt-get install -y sudo iproute2 iputils-ping
--------------------
ERROR: failed to solve: phusion/baseimage:master: docker.io/phusion/baseimage:master: not found

dominicdoty, 2023-09-20:
If you get stuck here, you need to change your Dockerfile to point to a specific image like:

`FROM phusion/baseimage:jammy-1.0.1`

Look for tags: [Phusion Dockerhub](https://registry.hub.docker.com/r/phusion/baseimage/tags)

See: [phusion issue for master tag](https://github.com/phusion/baseimage-docker/issues/625)

EBADBEEF, 2024-07-02:
I suggest using `FROM phusion/baseimage:bionic-1.0.0` (Ubuntu-18.04) otherwise the script was not successful (looks the same as https://github.com/ct-Open-Source/tuya-convert/issues/1022). I was able to upgrade two [Bestek MRJ10011](https://templates.blakadder.com/bestek_MRJ10011.html) using `docker compose`. I used my desktop linux with an Intel AX200 (iwlwifi driver) wifi card. I had to change the docker-compose.yml to specify `WLAN=wlp9s0` for my system.