ct-Open-Source/tuya-convert #1076: Failed to flash AWP07L Socket

PascalTrm, 2022-12-14:
**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

**Please remember to close your issue when the problem has been addressed!**

Hi all, 

I just tried to flash my Avatar Mini Smart WIFI Plug (AWP07L) Socket that I had laying around using a Rasberry Pi 4.
Sadly the process always failed and I can't find a log message pointing to the problem.

The Socket seems to connect to the WIFI for updating but then the plug just turns off. 
Here are the logs: 

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/10228709/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/10228712/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/10228714/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/10228716/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/10228719/smarthack-wifi.log)

TrustedFloppa, 2022-12-17:
Here is what i found in logs
[image]
Try connecting vtrust-flash to other device like laptop or smt else

PascalTrm, 2022-12-17:
Those logs happened because my Phone kept on disconnecting from the WiFi when going to sleep. I also tried it with my PC making sure it doesn't disconnect, but sadly the flash Process still fails.

TrustedFloppa, 2022-12-17:
@PascalTrm,
Is the plug going into AP mode? (slow blink)

PascalTrm, 2023-01-09:
Hi @TrustedFloppa, sorry for the late reply, it has been busy. No, it doesn't look like it's going into AP mode. The light is blinking fast as expected.

TrustedFloppa, 2023-01-09:
Try using other device with WiFi adapter

ebuford, 2023-02-02:
II had a very similar issue that I just resolved by reflashing my Raspberry Pi with the Legacy version (Buster) of Raspbian and then setting up Tuya-Convert on this version.
I know it's a bit of a hassle but the folks on digiblurDIY discord helped me through it.