ct-Open-Source/tuya-convert #1078: Does not work

hackintoshlover, 2023-01-08:
Hi,
I am trying to install and run tuya convert but it does not work.
I am using raspberry pi 4b+ 2gb ram 16gb sdcard freshly installed raspbian buster image, connected over lan.
Cloned the repository, installed prerequisite but when i run ./start_flash.sh nothing happens.
Kindly refer attached image, kindly help.
[image]

JosefPRajmon, 2023-01-09:
type "yes"

hackintoshlover, 2023-01-10:
Thanks, It worked.
So dumb of me.
It says i have got new firmware so the setup won't be able to work.
If we want a workaround for the new tuya patch what are the steps that we will need to follow?
I know it can be done over serial but nothing beats wireless.

tedztar, 2023-07-04:
As far as I have found I haven't seen a work around, and I'm in the same boat as you. my firmware is too new.

samip5, 2023-09-16:
> Thanks, It worked. So dumb of me. It says i have got new firmware so the setup won't be able to work. If we want a workaround for the new tuya patch what are the steps that we will need to follow? I know it can be done over serial but nothing beats wireless.

I was able to just ask them to push firmware update to a version that allows conversion.

hackintoshlover, 2023-09-16:
How exactly?

samip5, 2023-09-16:
> How exactly?

Contact <globalservice@gosund.com> with your device's virtual ID from the app, and specify in the email in which APP it is in.

FAIR WARNING: If your plug is too new, it cannot be downgraded unless you flash it via the pins.