ct-Open-Source/tuya-convert #1081: Gosund EP2 not a issue but a solve for using this tool

TrustedFloppa, 2023-01-17:
I messaged gosund support (globalservice@gosund.com)
And I asked for 1.0.6 firmware that works with this tool
They replied that they will send me a update then my sockets got updated to 1.0.6 in 3 days by gosund support (sadly only EP2)
I hope I helped  (I don't know if someone already mentioned it)

jpdribbler, 2023-02-24:
I am really happy with gosound. They have been friendly and fast. Thank you Chinese friends!

Exactly 48 hours and i could update my GOSUND EP2 to 1.06.
Tomorrow i will try tuy-convert... :)

TrustedFloppa, 2023-02-24:
I updated mine to tasmota recently

hookro, 2023-03-16:
Respond from Gosund. They seem friendly. However I managed to update SP211 without them.

[image]

TrustedFloppa, 2023-03-17:
@hookro 1.0.8? i tought they can be updated max to 1.0.6

zicheee, 2023-04-24:
I bought 6 EP2 in total and all where updated to 1.0.8.
With a Raspberry Pi 4, Bullseye Lite and this software I managed to install tasmota on all of them.

TrustedFloppa, 2023-04-24:
i bricked one but i got it replaced, because I used lastest one from tasmota-lite.bin and it just bricked it.

TrustedFloppa, 2023-05-06:
They replaced it in store because they had 0 clue it's Tasmota flashed

Fahl-Design, 2023-05-21:
Got 12 of them and can confirm you get version 1.0.8 if you message support.

I used my notebook and the docker version (and an old edimax wifi stick) to flash OTA (included tasmoto.bin and later updated to newest in web-ui)

Profile:
```json
{"NAME":"Gosund EP2","GPIO":[576,1,320,1,2656,2720,0,0,2624,32,0,224,0,0],"FLAG":0,"BASE":45}
```