ct-Open-Source/tuya-convert #1083: Avatar ‎AWP16L Error

tzockt, 2023-01-24:
### Before opening a new issue, please read the following

**Are you reporting a successful flash?**
_Please add it to the wiki instead!_

**Do you have a question about configuring your newly flashed device?**
_Please ask in your alternative firmware's support channel instead!_

**Did you have issues getting a device to flash?**
_Please be sure to include your device **name, model, and firmware version** as well as your **logs**!_

Hi all,

I just tried to flash my Avatar ‎AWP16L Socket that I had laying around using a Rasberry Pi 4.
Sadly the process always failed and I can't find a log message pointing to the problem.
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/10494840/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/10494841/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/10494842/smarthack-wifi.log)
[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/10494843/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/10494844/smarthack-psk.log)

ebuford, 2023-02-02:
I had a very similar issue that I just resolved by reflashing my Raspberry Pi with the Legacy version (Buster) of Raspbian and then setting up Tuya-Convert on this version. 
I know it's a bit of a hassle but the folks on digiblurDIY discord helped me through it.

tzockt, 2023-02-02:
With buster it has now worked but my firmware is unfortunately too new :(
```Attempting to diagnose the issue...
Your device's firmware is too new.
Tuya patched the PSK vulnerability that we use to establish a connection.
You might still be able to flash this device over serial.
For more information and to follow progress on solving this issue see:
https://github.com/ct-Open-Source/tuya-convert/wiki/Collaboration-document-for-PSK-Identity-02```

ebuford, 2023-02-02:
Well that is sad - but at least you got passed the error!