ct-Open-Source/tuya-convert #1084: Unable tto flash BW-SHP2 V2.3

evertos, 2023-01-27:
I'm unable to flash: Blitzworlf BW-SHP2 V2.3
Hopefully, I'm doing something wrong but it looks like the device is responding as it stops blicking with the led.

Thank you verry much for working on this project!

Full Output, using a Raspberry pi3b to run it:

```
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Timed out while waiting for the device to (re)connect
```

ebuford, 2023-02-02:
In the Tuya-Convert folder is a scripts folder - cd into that folder and copy out all your log files and post them. It will help!
I had a very similar issue that I just resolved by reflashing my Raspberry Pi with the Legacy version (Buster) of Raspbian and then setting up Tuya-Convert on this version.
Let me know if this helps.
Again - posting your logs will also help us help you.

evertos, 2023-02-04:
Awsome! Indeed did install Buster and that is working. Still got some problems but I got a Tasmota webinterface so that should be fine!

EHoegee, 2023-02-20:
Well, this did solve my issue, some more information in order to add it to openhab.
I needed to downgrade it in order to enable discovery.
https://community.openhab.org/t/how-set-up-the-native-tasmota-discovery/139112/13

evertos, 2023-02-20:
EHoegee, is my second account, so this is solved.
install Buster and that is working :D 
@ebuford I'm pretty new to tasmota, do we want to keep this issue open so they can check why it did not work on the new RP image? or do you know if there is an open issue on this?

closed 2023-02-20 by evertos

ebuford, 2023-02-20:
> 

If you have worked through the issue, I think it it should be closed. 
Glad you got it working!