ct-Open-Source/tuya-convert #1086: Gosund EP2 firmware 1.0.6 flashing tasmota-minimal 12.3.1 - successfully flashed, but dead afterwards

rudelm, 2023-02-09:
Hi,

I've used a Raspberry Pi3 and its internal Wifi on Raspbian release 2022-09-22. The installation process of tuya-convert went fine and without problems. I've replaced the original tasmota.bin with https://github.com/arendst/Tasmota/releases/download/v12.3.1/tasmota-lite.bin which is the latest version 12.3.1 and came from version 1.0.4 which was updated by the Gosund support to 1.0.6 so I could flash Tasmota.

I've connected during the process with my iPhone to the VTRUST Wifi and was shown a wifi portal message that I'm connected, without being forwarded from that page.

I was able to get my two plugs connected to the pi and flashed successfully the image. It even created the backup images fine. But after a while no new Wifi network was visible. I've plugged them in and out and pressed the power button, but I was unable to get any reaction. The only thing that happens is during plugging it in is a short green LED flash. Besides that they appear both to be dead.

What are my options now? Cracking the case open to solder a direct connection to the board, trying a direct recovery?

Here's the output from my flash process:

`TUYA-CONVERT

https://github.com/ct-Open-Source/tuya-convert
TUYA-CONVERT was developed by Michael Steigerwald from the IT security company VTRUST (https://www.vtrust.de/) in collaboration with the techjournalists Merlin Schumacher, Pina Merkert, Andrijan Moecker and Jan Mahn at c't Magazine. (https://www.ct.de/)

======================================================
PLEASE READ THIS CAREFULLY!
======================================================
TUYA-CONVERT creates a fake update server environment for ESP8266/85 based tuya devices. It enables you to backup your devices firmware and upload an alternative one (e.g. ESPEasy, Tasmota, Espurna) without the need to open the device and solder a serial connection (OTA, Over-the-air).
Please make sure that you understand the consequences of flashing an alternative firmware, since you might lose functionality!

Flashing an alternative firmware can cause unexpected device behavior and/or render the device unusable. Be aware that you do use this software at YOUR OWN RISK! Please acknowledge that VTRUST and c't Magazine (or Heise Medien GmbH & Co. KG) CAN NOT be held accountable for ANY DAMAGE or LOSS OF FUNCTIONALITY by typing yes + Enter

yes
Checking for network interface wlan0... Found.
Checking UDP port 53... Occupied by dnsmasq with PID 2574.
Port 53 is needed to resolve DNS queries
Do you wish to terminate dnsmasq? [y/N] y
Attempting to stop dnsmasq.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Occupied by mosquitto with PID 2956.
Port 1883 is needed to run MQTT
Do you wish to terminate mosquitto? [y/N] y
Attempting to stop mosquitto.service
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.....
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.............................................................................................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  61428      0  0:00:17  0:00:17 --:--:-- 28953
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 5df024
MAC: 48:55:19:5D:F0:24
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 14325e
FlashChipRealSize: 1024K
Active Userspace: user1 0x01000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Flashed http://10.42.42.1/files/tasmota.bin successfully in 22332ms, rebooting...
Look for a tasmota-xxxx SSID to which you can connect and configure
Be sure to configure your device for proper function!

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] y
======================================================

IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
...............
SmartConfig complete.
Resending SmartConfig Packets
........................................................................................................
IoT-device is online with ip 10.42.42.42
Fetching firmware backup
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 1024k  100 1024k    0     0  61954      0  0:00:16  0:00:16 --:--:-- 28379
======================================================
Getting Info from IoT-device
VTRUST-FLASH 1.5
(c) VTRUST GMBH https://www.vtrust.de/35c3/
READ FLASH: http://10.42.42.42/backup
ChipID: 50ee54
MAC: 48:55:19:50:EE:54
BootVersion: 7
BootMode: normal
FlashMode: 1M DOUT @ 40MHz
FlashChipId: 14325e
FlashChipRealSize: 1024K
Active Userspace: user1 0x01000
======================================================
Ready to flash third party firmware!

For your convenience, the following firmware images are already included in this repository:
  Tasmota v8.1.0.2 (wifiman)
  ESPurna 1.13.5 (base)

You can also provide your own image by placing it in the /files directory
Please ensure the firmware fits the device and includes the bootloader
MAXIMUM SIZE IS 512KB

Available options:
  0) return to stock
  1) flash espurna.bin
  2) flash tasmota.bin
  q) quit; do nothing
Please select 0-2: 2
Are you sure you want to flash tasmota.bin? This is the point of no return [y/N] y
Attempting to flash tasmota.bin, this may take a few seconds...
Flashed http://10.42.42.1/files/tasmota.bin successfully in 21907ms, rebooting...
Look for a tasmota-xxxx SSID to which you can connect and configure
Be sure to configure your device for proper function!

HAVE FUN!
======================================================
Do you want to flash another device? [y/N] n
======================================================
Cleaning up...
Closing AP
Exiting...`

rudelm, 2023-02-09:
Looks like this is almost "normal" for this device https://github.com/arendst/Tasmota/discussions/10350 regardless if I use the latest Tasmota or the one that's part of tuya-convert. A larger amount of plugs ends up bricked after OTA updates. Cracking the case open seems very destructive so I'll guess these plugs are now broken.

rudelm, 2023-02-13:
OK, I've got two other EP2s converted to 1.06 which I will try with the tasmota image part of this repo.

Are there any known limitations with newer tasmota versions that would hinder a direct flash to the latest version? this might justify a small note in the readme.md to not update the image file and to do later update attempts manually via Tasmota UI.

rudelm, 2023-02-13:
Maybe the tasmota image I've used doesn't fulfill these requirements:

```
Binary requirements:

full binary including first-stage bootloader (tested with Arduino eboot and Open-RTOS rBoot)
maximum filesize 512KB for first flash
```

I could now update the two EP2 plugs with the default image from the repo.

closed 2023-02-13 by rudelm

TrustedFloppa, 2023-02-13:
I have EP2 sockets with flashed tasmota using tuya-convert and I updated from 9.t.0 to lastest one

rudelm, 2023-02-14:
yeah was also able to flash it that way. Tried first the lite image, as the tuya-convert was also a lite image and tried then the regular sized tasmota image. That's how I was used to update my existing Gosund SP1.