ct-Open-Source/tuya-convert #1091: I add image from valve. Also issue is present even though I did not connect it ever to Tuya.

pritam2016bakshi, 2023-03-05:
I add image from valve. Also issue is present even though I did not connect it ever to Tuya.
https://templates.blakadder.com/jinvoo_SM-PW713.html
[backup_20200711_210602.zip](https://github.com/ct-Open-Source/tuya-convert/files/4907683/backup_20200711_210602.zip)

_Originally posted by @AdamWeglarz in https://github.com/ct-Open-Source/tuya-convert/issues/483#issuecomment-657138956_

Firmware file is correpted, so upload new one.

kueblc, 2023-03-05:
Hi @pritam2016bakshi,

I'm not clear on what you're asking, although it doesn't seem to be related to an issue with the tuya-convert software.

closed 2023-03-05 by kueblc

kueblc, 2023-03-07:
@pritam2016bakshi I see you cross posting a request for firmware on multiple issues.

I'm gathering that you tried to flash someone else's firmware backup onto your device and it did not work so you assumed that the backup was corrupted.

Please note that you will not be able to use a device that has been flashed with the stock firmware from another device. The stock firmware contains unique features that restrict its operation to the original device. You would need to do a lot of additional work to try and "trick" it into running on a different device.

pritam2016bakshi, 2023-03-07:
Sir, Can you teach me this trick?