ct-Open-Source/tuya-convert #1096: Potential Wireless Flash for EKAZA Smart Plug 16A - TYWE2S - MCU V1.1.7?

camnop, 2023-06-15:
Hello everyone,

I've been having a bit of a head-scratcher lately. I've got this **EKAZA Smart Plug, a 16A model with a TYWE2S chip, rocking firmware MCU V1.1.7**. I've been trying to flash it with esphome, but there's a bit of a hiccup - the firmware seems to be not compatible with the tools I usually use, [LibreTiny and Tuya-CloudCutter.](https://github.com/kuba2k2/libretiny/issues/138#issue-1757865231)

And now, the big question mark: Is there a way to flash this bad boy wirelessly, without resorting to surgery (aka opening the box)? Or if I have to go the wired route, is it possible with serial mode using a cable? And if so, do I need to connect any GPIO to the ground?

I'm super grateful for any insight you can provide. And if there's any additional info you need about the device, just let me know! Here's the rundown:

- Device: EKAZA Smart Plug 16A
- Chip: TYWE2S
- Firmware Version: MCU V1.1.7

Thanks a ton and looking forward to your thoughts!

[image]
[image]
[image]
[image]

jkwim, 2023-06-25:
See wether this helps:

Apparently Tuya has made changes to the TYWE2S module itself to expose GPIO0 in place of AD pin in some devices.

https://www.elektroda.com/rtvforum/topic3974457.html

jkwim, 2023-06-25:
Also post a picture of the backside of the module.

[image]

There might be a test pad at the back of it for GPIO0.

jkwim, 2023-06-25:
https://github.com/ct-Open-Source/tuya-convert/issues/862

ctrl-freak, 2023-07-12:
> #862

@jkwim Thanks for the issue link and your photos; I couldn't find GPIO0 because I was reading that test pad as "100", as in 'one hundred'...