ct-Open-Source/tuya-convert #1101: STAY AWAY from Gosund plugs

samip5, 2023-09-16:
I got confirmation from their support that most of their plugs are already upgraded versions and cannot be downgraded.

For reference:
```
From: Mufi (Support) <globalservice@gosund.com>
To: Me <censored>@gmail.com
Subject: Re: Firmware request

Hi Skyler,
Thank you for your email.
Most of our plugs are already upgraded and they no longer support flashing to Tasmota.
We are sorry to inform you that upon checking, your smart plug is an upgraded version. Therefore, the firmware cannot be updated to v1.0.8.
We are hoping for your understanding.
```

mlinko6, 2023-11-11:
Which Version of Tasmota do you have?

KR
Rok

samip5, 2023-11-11:
> Which Version of Tasmota do you have?
> 
> KR
> Rok

None? It comes with Tuya.

mlinko6, 2023-11-11:
8.2.0 - 6x on and off restart solved the issue.

KR
Rok

samip5, 2023-11-12:
> Sorry I mean’t Gosund version

Main Module lists version as v1.1.15 and Other type: v1.0.7 so expect every single one of these to be non-downgradeable if you're currently thinking about them.

mlinko6, 2023-11-12:
I've just flashed Gosund EP2 smart plugs with the firmware version: 1.0.8 4 with Tuya converter installed on Raspberry Pi4 - i had no Issues with flashing.

samip5, 2023-11-12:
> I've just flashed Gosund EP2 smart plugs with the firmware version: 1.0.8 4 with Tuya converter installed on Raspberry Pi4 - i had no Issues with flashing.

That version is exactly the ONLY ONE which works, but most ones you can buy anymore are running newer than that, and thus will not work.

broyuken, 2023-11-29:
you should still be able to open them up and flash them

samip5, 2023-11-30:
> you should still be able to open them up and flash them

Doesn't matter, as I'm not going to open it.

TrustedFloppa, 2023-11-30:
if you get some old models from the store then you may have 1.0.4 firmware.(Probably tuya forced them to stop flashing EP2 to customers plugs) But there is also EP2 with a beken chip.
~~gosund ran out of esp chips??~~

TrustedFloppa, 2023-11-30:
so sad they are patched, i didn't get to flash tasmota on mine :(