ct-Open-Source/tuya-convert #1105: Deta Grid Connect Lights dead with vtrust-recovery SSID after flash

Inglan, 2023-09-25:
Hi, the other day I was flashing 2 [grid connect](https://www.bunnings.com.au/deta-grid-connect-smart-18w-1900lm-daylight-t8-tube_p0111511) lights with Tasmota. The process went well for one of the lights (though I _did_ have to configure tasmota), but the other will now not turn on, and it's SSID is vtrust-recovery. When I connect to the network, it disappears. I have seen people with the same issue, that say to ./start_flash.sh and then turn it on, but this does not work. It starts backing up the firmware but the transfer speed drops and then stops entirely.
Thank you
Inglan

kueblc, 2024-01-04:
Have you tried skipping the backup? You can just comment that section out in `start_flash.sh`

Inglan, 2024-01-09:
No not yet. Will try when I get the chance