ct-Open-Source/tuya-convert #1114: no AP "vtrus-flash" is opend - script is running without error

Tobbestoll, 2023-11-10:
Hello,

there is no AP like "vtrust-flash" opened. My wifi card is "wlan0". I tried different wifi cards also without any AP shown.
Did also "rfkill unblock wifi" and "killall wpa_supplicant" without any success.

The script gave me no error. It shows the message that I should connect with the AP and start the flashing process.

How can I get the necessary access point ?

linesquarecube, 2023-11-21:
Hey @Tobbestoll,

which Raspberrypi OS version did you use? I had the same problem and switching back from bookworm to bullseye did the trick for me. I've read the reason why it is not working with bookworm somewhere, but I can't remember the source.

Hopefully that helps.

Tobbestoll, 2023-11-21:
Thanks a lot. I used an old Pi+ with Pi OS Legacy lite
This combi does the hotspot.
Unfortunately the wifi plug does not contain an ESP :-)

closed 2024-01-04 by kueblc