ct-Open-Source/tuya-convert #1117: Gosund EP2 flashed with Tasmota - but entered wrong Wifi Name or Password and can't reach the device

mlinko6, 2023-11-11:
Dear AlI,

I've successfully flashed all 4 of my Gosund EP2 (SP111) but on one occasion I've probably made a typo in either Wifi Name or Password, so the Plug is not "reacting" at all. It didn't connect to the wifi as it also didn't reset the wifi back to "tasmota_xxxxxx" . 

Do you have any Tipps on what can I do to solve my Issue?

Will it help if I leave it without power for 3 or 4 days?

Any Tipp will be much appreciated.

Thx!

KR
Rok

Noschvie, 2023-11-11:
Have a loook to https://tasmota.github.io/docs/Device-Recovery/

mlinko6, 2023-11-11:
Thx it worked like a charm! I've tried it 2 days ago but maybe I was to hasty with 6x times in and out :-).

Thx

Noschvie!

bugsyb, 2024-07-04:
@mlinko6 , would you mind to share if any tricks needed to be done to flash EP2? I've failed to flash the SP111 - 3680W with tuya-convert and had to open them up, hence asking.

mlinko6, 2024-07-04:
You mean an non resposive one? I didn't have any problem with the normal tuya-convert procedure. I've done it on 6 Devices without a problem.

https://github.com/ct-Open-Source/tuya-convert

I did it with an Rapsberry Pi4