ct-Open-Source/tuya-convert #1121: Timed out while waiting for the device to (re)connect

ajvisser99, 2023-12-27:
Hello All,

I have been trying to flash some light bulbs using this, and seem to be running into some challenges. I looked through many other similar issues posted, but have not found one that seems to be exactly the same as mine.  I am running the most recent version of Pi OS on a pi 4b. 
I am getting the follow. 
======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Auto retry in 2s. .....................................................................
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
An ESP82xx based device connected according to your wifi log.
If this is the device you are trying to flash, another issue may be preventing it from
Otherwise, it could be that the device does not use an ESP82xx or it did not connect.
For additional information, check the *.log files inside the scripts folder.
Please include these logs when opening a new issue on our GitHub issue tracker.

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/13780107/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/13780108/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/13780109/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/13780110/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/13780111/smarthack-wifi.log)

Appreciate the help in advance :)

kueblc, 2024-01-04:
I see that this is a first generation Tuya device, there is hope yet!

As shown in your web log, it does indeed connect successfully and issue the upgrade command. At that point, the firmware loader is never requested.

I have seen this reported before. It may mean we are missing something in our mocked API responses. It varies between device models and firmware versions. More research is required.

@velkrosmaak without your logs, I can't confirm if your issue is the same or not. Repeated smartconfig is a catch all symptom.

lkwilson, 2024-04-09:
I believe I am seeing the same thing with Techin SP10 plugs

[smarthack-mqtt.log](https://github.com/ct-Open-Source/tuya-convert/files/14913178/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/ct-Open-Source/tuya-convert/files/14913179/smarthack-psk.log)
[smarthack-udp.log](https://github.com/ct-Open-Source/tuya-convert/files/14913180/smarthack-udp.log)
[smarthack-web.log](https://github.com/ct-Open-Source/tuya-convert/files/14913181/smarthack-web.log)
[smarthack-wifi.log](https://github.com/ct-Open-Source/tuya-convert/files/14913182/smarthack-wifi.log)

pikim, 2025-02-17:
I think I may have the same problem. Is there any solution, yet?

[smarthack-mqtt.log](https://github.com/user-attachments/files/18827710/smarthack-mqtt.log)
[smarthack-psk.log](https://github.com/user-attachments/files/18827713/smarthack-psk.log)
[smarthack-udp.log](https://github.com/user-attachments/files/18827714/smarthack-udp.log)
[smarthack-web.log](https://github.com/user-attachments/files/18827711/smarthack-web.log)
[smarthack-wifi.log](https://github.com/user-attachments/files/18827712/smarthack-wifi.log)